balena-io/etcher #4286: Cannot be packaged for Nix as package-lock.json does not exist

myclevorname, 2024-07-29:
- **Etcher version:** 1.19.22
- **Operating system and architecture:** NixOS unstable x86_64-linux
- **Image flashed:** N/A
- **What do you think should have happened:** There should be a package-lock.json file, allowing reproducible builds
- **What happened:** There is no package-lock.json file, meaning builds are not reproducible
- **Do you see any meaningful error information in the DevTools?** N/A

I wanted to see if I could package this for Nix, but the lack of `package-lock.json` means the only other option is to add in the file as a patch.

lurch, 2024-07-29:
Does `npm-shrinkwrap.json` do a similar thing?

myclevorname, 2024-07-29:
I don't know. On Discord, I had been reminded that dd exists, so I don't feel a need to package this.

closed 2024-07-29 by myclevorname