balena-io/etcher #4287: Blue screen when I start flashing

owendswang, 2024-07-29:
- **Etcher version:** 1.19.21
- **Operating system and architecture:** Windows 10 x64 22H2
- **Image flashed:** https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.2.1/ubuntu-24.04-preinstalled-server-arm64-rock-5b.img.xz
- **What do you think should have happened:** 
1. plugin sd card into usb card reader.
2. plugin card reader to laptop
3. start balenaEtcher
4. select the image
5. select sd card
6. flash
7. done
- **What happened:**
1. plugin sd card into usb card reader.
2. plugin card reader to laptop
3. start balenaEtcher
4. select the image
5. select sd card
6. flash
7. BLUE SCREEN
- **Do you see any meaningful error information in the DevTools?** 

Can't cuz blue screen.

[image]

lurch, 2024-07-29:
It might be a driver-problem with your usb card reader?

owendswang, 2024-07-29:
It's just a USB card reader for plug-and-use. No drivers were provided. Windows recognize it automatically.

owendswang, 2024-07-30:
This problem doesn't exist in Ubuntu Desktop 22.04.

owendswang, 2024-08-01:
It usually happens the second time when I flash the SD card after Windows boots up.