balena-io/etcher #4290: Mac OS version supported by Etcher 1.19.21?

olderccc, 2024-08-01:
- **Etcher version: 1.19.21 ** 
- **Operating system and architecture: MacOS 10.13.6 High Sierra i386** 
- **Image flashed: won't install** 
- **What do you think should have happened: requirements were met , should have installed** 
- **What happened: MacOS 10.13.6 is the requirement listed on GitHub page, but installer says I need MacOS 10.15 or later **
[image]
[image]
 
- **Do you see any meaningful error information in the DevTools? N/A **

armdn, 2024-08-20:
Same thing on macOS 10.14.6.

gillcar, 2024-09-06:
I use version 1.18.8 on my 2009-2010 Mac with El Capitain V10.11.6 without problem.
The lastest versions don't work on it