balena-io/etcher #4291: Unable to flash windows

mahe8696, 2024-08-03:
please add support for windows

silas-wr, 2024-08-05:
Can you please describe your problem more in-depth @mahe8696? Did it give you an error? Did it format the USB wrong? I looked in the release page and I found the EXE, as well as a ZIP file, both for Windows. Thank you!

lurch, 2024-08-06:
My assumption is that this is referring to https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable ?

SuleymanCommits, 2024-08-06:
no, they want the ability to flash Windows ISOs using Etcher

SuleymanCommits, 2024-08-06:
> Can you please describe your problem more in-depth @mahe8696? Did it give you an error? Did it format the USB wrong? I looked in the release page and I found the EXE, as well as a ZIP file, both for Windows. Thank you!

they want to flash Windows ISOs using Etcher

enessmr, 2024-08-14:
Windows Requires special imaging to be bootable. If use windows, Use RUFUS. Linux? https://linuxiac.com/how-to-create-a-windows-bootable-usb-on-linux/