balena-io/etcher #4294: Stuck at "Flash from file"

Gameking1happy, 2024-08-07:
- **Etcher version: 1.19.21** 
- **Operating system and architecture: Windows 10, HP Pavilion 17-e110nr Notebook PC** 
- **Image flashed: linuxmint-22-cinnamon-64bit** 
- **What do you think should have happened: After selecting file to flash from something happens.** 
- **What happened: After selecting file to flash from the "Flash from file" button is grayed out and nothing is shown to be happening.**
- **Do you see any meaningful error information in the DevTools?**
In Application tab it says:
App Manifest
(unknown)
Errors and warnings
Richer PWA Install UI won't be available on desktop. Please add at least one screenshot with the form_factor set to wide.
Richer PWA Install UI won't be available on mobile. Please add at least one screenshot for which form_factor is not set or set to a value other than wide.
Most operating systems require square icons. Please include at least one square icon in the array.

NikitaSavc, 2024-08-07:
+1
Confirm this!

jiri-matejka, 2024-08-10:
+1 here, on Fedora 39, installed Etcher 1.19.21 from RPM file

skols, 2024-08-13:
+1 on this issue. Fedora 40, running 1.19.22 AppImage.

dtutberidze, 2024-08-16:
+1 too, Windows 10, balenaEtcher-1.19.21

lurch, 2024-08-16:
See #4150

rickhat805, 2024-08-30:
+1 also, running on  Ubuntu 24.04.01, balena-etcher-19.21, trying to grab the Ubuntu-24-04-01 iso. same error message as shown above by NikitaSavc.