balena-io/etcher #4295: DEB Packages depend on packages no longer available in Ubuntu 24.04

gdevenyi, 2024-08-10:
- **Etcher version:** 1.19.21
- **Operating system and architecture:** Kubuntu 24.04 x86_64
- **Image flashed:** N/A
- **What do you think should have happened:** Should install 
- **What happened:**

The deb file has the following metdata:
```
dpkg -I balena-etcher_1.19.21_amd64.deb
 new Debian package, version 2.0.
 size 110310096 bytes: control archive=1156 bytes.
    1262 bytes,    16 lines      control
     357 bytes,    11 lines   *  postinst             #!/bin/bash
 Package: balena-etcher
 Version: 1.19.21
 Section: utils
 Priority: optional
 Architecture: amd64
 Depends: libgtk-3-0, libnotify4, libnss3, libxtst6, xdg-utils, libatspi2.0-0, libdrm2, libgbm1, libxcb-dri3-0, kde-cli-tools | kde-runtime | trash-cli | libglib2.0-bin | gvfs-bin, gconf-service, gconf2, libasound2, libatk1.0-0, libc6, libcairo2, libcups2, libdbus-1-3, libexpat1, libfontconfig1, libfreetype6, libgcc1, libgconf-2-4, libgdk-pixbuf2.0-0, libglib2.0-0, liblzma5, libnspr4, libpango1.0-0 | libpango-1.0-0, libstdc++6, libx11-6, libxcomposite1, libxcursor1, libxdamage1, libxext6, libxfixes3, libxi6, libxrandr2, libxrender1, libxss1, polkit-1-auth-agent | policykit-1-gnome | polkit-kde-1
 Recommends: pulseaudio | libasound2
 Suggests: gir1.2-gnomekeyring-1.0, libgnome-keyring0, lsb-release
 Installed-Size: 396626
 Maintainer: Balena Ltd. <hello@balena.io>
 Homepage: https://github.com/balena-io/etcher
 Description: Flash OS images to SD cards and USB drives, safely and easily.
  Etcher is a powerful OS image flasher built with web technologies to ensure
  flashing an SDCard or USB drive is a pleasant and safe experience. It protects
  you from accidentally writing to your hard-drives, ensures every byte of data
  was written correctly and much more.
```

Attempting to install:
```
$ sudo apt-get install ./balena-etcher_1.19.21_amd64.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'balena-etcher' instead of './balena-etcher_1.19.21_amd64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help resolve the situation:

The following packages have unmet dependencies:
 balena-etcher : Depends: gconf-service but it is not installable
                 Depends: gconf2 but it is not installable
                 Depends: libgconf-2-4 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

All of those packages are depreciated and not available after 22.04:
gconf-service https://packages.ubuntu.com/search?suite=all&section=all&arch=any&keywords=gconf-service&searchon=names
gconf2 https://packages.ubuntu.com/search?suite=all&section=all&arch=any&keywords=gconf2&searchon=names
libgconf-2-4 https://packages.ubuntu.com/search?suite=all&section=all&arch=any&keywords=libgconf-2-4+&searchon=names

gdevenyi, 2024-08-10:
Follwing the instructions at:
https://forums.linuxmint.com/viewtopic.php?t=35136

I was able to unpack the deb file, and remove the missing dependencies, and repack the deb file, and it installs and runs under 24.04

This leads me to believe a simple fix may be to correct the dependencies:
https://github.com/balena-io/etcher/blob/master/package.json

When is the last time these were audited for actual need?

therealziggy, 2024-08-11:
> Follwing the instructions at: https://forums.linuxmint.com/viewtopic.php?t=35136
>
> I was able to unpack the deb file, and remove the missing dependencies, and repack the deb file, and it installs and [...]

thanks!

Angular-Angel, 2024-08-19:
Well, I tried editing the debian package to remove the unnecessary dependencies, and while I was able to install it, it still won't run:
```
angle@Sager:~/Downloads$ balena-etcher
[59200:0819/132437.295296:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /usr/lib/balena-etcher/chrome-sandbox is owned by root and has mode 4755.
Trace/breakpoint trap (core dumped)
```
Running `sudo chmod 4755 /usr/lib/balena-etcher/chrome-sandbox` did not help. possibly because the file is still owned by my user, rather than root? But I had it as root when I tried using the RPM package via alien earlier, and it didn't like that either... :/

gdevenyi, 2024-08-19:
Your existing install contamination is like the cause here.

Jemo121, 2024-08-25:
Minor issue encountered while installing Etcher version: 1.19.21 in [Kali Linux](https://www.kali.org/docs/policy/kali-linux-relationship-with-debian/) 2024.2, after:

`$ sudo apt-get install ./balena-etcher_1.19.21_amd64.deb`

During installation received:

`Warning: program compiled against libxml 212 using older 209`

Is there any way to resolve this?

lurch, 2026-05-09:
Fixed by v2.1.5 ?