balena-io/etcher #4296: Error Spawning Child Process

MasterLuma, 2024-08-13:
- **Etcher version: Latest as of this post 1.19.21** 
- **Operating system and architecture: Win10 Home 22H2 64Bit x64** 
- **Image flashed: Attempted to clone 1TB SATA SSD to 4TB SATA SSD** 
- **What do you think should have happened:** Open program as Admin, choose source drive, select target drive, then start process and complete successfully.
- **What happened: Error Spawning Child Process**
- **Do you see any meaningful error information in the DevTools?** 

This seems to be a popular issue right now. I have tried many different methods of connecting the drives and running the program, all to no avail. I get the same issue with other types of data storage devices. Balena is completely unusable right now

cuflyer, 2024-08-21:
I have the same problem. Many sd's, a couple of card adapters, 2 different computers.
Cannot flash an image file.

maxdiluca, 2024-08-22:
Tried on two computers with macOS with both i7 and M3, different SD card and readers, and I keep receiving the same error

cuflyer, 2024-08-22:
I just ended up using another utility, Rufus. Worked great.

Crys7a1Grain, 2024-09-04:
Same problem on Win10