balena-io/etcher #4297: Any balenaEtcher version for Windows XP?

enessmr, 2024-08-14:
- **Etcher version:Not downloaded yet, want a XP version, No issues in 11** 
- **Operating system and architecture: amd64 X86 xp x64 11** 
- **Image flashed: lots of** 
- **What do you think should have happened: Write system req** 
- **What happened: cannot find for xp version**
- **Do you see any meaningful error information in the DevTools?**

enessmr, 2024-08-14:
Chrome in Windows 11. Making a POWIS XP2ESD image.

enessmr, 2024-08-14:
[image]

lurch, 2024-08-14:
https://github.com/balena-io/etcher?tab=readme-ov-file#supported-operating-systems says "Windows 10 and later; Intel 64-bit.".

enessmr, 2024-08-14:
exe file so i can enter text

enessmr, 2024-08-14:
or give the Wayback machine date so i can do it

enessmr, 2024-08-14:
does not exist

enessmr, 2024-08-14:
wayback machine

enessmr, 2024-08-14:
proof (sorry i can't did too good i didnt scale it to make better)
[image]