balena-io/etcher #4305: Cannot Install Balena Etcher on Ubuntu 24.04

littledog33, 2024-08-25:
- 1.19 or 1.18; tried both 
- Ubuntu Studio 24.04
- Tried executable from zip file, deb file with apt, and 32-bit AppImage
- Should have installed as in the past.
- Error Message:
The following packages have unmet dependencies:
 balena-etcher-electron : Depends: gconf2 but it is not installable
                          Depends: gconf-service but it is not installable
                          Depends: libgconf-2-4 but it is not installable
- Gconf has been deprecated from Debian upstream, and from Ubuntu 24.04. I tried installing the 32-bit AppImage, but the system could not find that file, even with the correct file name. 
- Is there a 64-bit Linux AppImage available?

dsssssssss9, 2024-08-25:
also having problems with Etcher on 24.04
Fresh install

if i double click the appimage nothing happens - other appimages work 

if i run the appimage from the terminal  i get.....

> [78996:0825/201952.035012:FATAL:setuid_sandbox_host.cc(157)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_balenaoD2g34/chrome-sandbox is owned by root and has mode 4755.
/tmp/.mount_balenaoD2g34/balena-etcher: line 10: 78996 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher.bin "$@"

if i run this with --no-sanbox switch i t seems to start but i get  warning in the terminal window....

`balena-etcher.bin:79032): dbind-WARNING **: 20:21:24.221: Couldn't connect to accessibility bus: Failed to connect to socket /root/.cache/at-spi/bus_0: Permission denied
`

any help greatly received

lewinfox, 2024-08-28:
:+1: for OP - same error message about unavailable dependencies.

wzk0, 2024-08-30:
I have tried `balenaEtcher-linux-x64-1.19.21.zip` and it can run very well.

littledog33, 2024-08-30:
How did you install it? On what distribution?

wzk0, 2024-08-30:
Ubuntu 24.04, just unzip it and run the file.

littledog33, 2024-08-30:
Thanks, but on Ubuntu Studio Linux 24.04, it wouldn't run.

littledog33, 2024-08-30:
Thanks for responding! 
  
I re-downloaded and extracted the zip file. I then ran chmod 4755 on balenaEtcher, but when I try to run it as root, I get an error message: 
  
 
error while loading shared libraries: libffmpeg.so: cannot open shared object file: No such file or directory 
  
So I installed ffmpeg, just to be sure. Same result, and I cannot find libffmpeg.so with an internet search.

wzk0, 2024-08-30:
You are welcome and just try run this without root permission!

 

    
    
   
    
       
 
 
 
   
 
 
 
 
 
 
 
 
 
 &amp;nbsp;  
 
 
 
 
 
 &amp;nbsp; I have tried balenaEtcher-linux-x64-1.19.21.zip and it can run very well. —

littledog33, 2024-08-31:
Finally got it to work! When I tried running it as my own username and not root, I got an error message to make sure that 
  
/home/mikeb/Downloads/balena-Etcher-linux-x64/chrome-sandbox 

was owned by root and had permissions 4755. I did those changes in the terminal, and then it finally ran.  
  
Thanks for your help!

wzk0, 2024-08-31:
I am very happy for singing this! You are welcome!

 

    
 
    
       
 
 
 
   
 
 
 
 
 
 
 
 
 
 Thanks for responding!  
 &amp;nbsp;  
 &amp;nbsp;  
 
 &amp;nbsp;  
 
 
 
 
 
 &amp;nbsp; Ubuntu 24.04, just unzip it and run the file.  
 
 
 
 ---Original---  
 
 
 
 
 
 &amp;amp;nbsp;  
 
 
 
 
 
 &amp;amp;nbsp; I have tried balenaEtcher-linux-x64-1.19.21.zip and it can run very well. —

fili, 2024-09-03:
Trying to install the .deb on 24.04.1 LTS using command line or software install leads to the following error:

```
$ sudo dpkg -i balena-etcher_1.19.22_amd64.deb 
Selecting previously unselected package balena-etcher.
(Reading database ... 431045 files and directories currently installed.)
Preparing to unpack balena-etcher_1.19.22_amd64.deb ...
Unpacking balena-etcher (1.19.22) ...
dpkg: dependency problems prevent configuration of balena-etcher:
 balena-etcher depends on gconf-service; however:
  Package gconf-service is not installed.
 balena-etcher depends on gconf2; however:
  Package gconf2 is not installed.
 balena-etcher depends on libgconf-2-4; however:
  Package libgconf-2-4 is not installed.

dpkg: error processing package balena-etcher (--install):
 dependency problems - leaving unconfigured
Processing triggers for gnome-menus (3.36.0-1.1ubuntu3) ...
Processing triggers for desktop-file-utils (0.27-2build1) ...
Processing triggers for mailcap (3.70+nmu1ubuntu1) ...
Errors were encountered while processing:
 balena-etcher
```

Trying to install these packages leads to the following errors:

```
E: Package 'gconf-service' has no installation candidate
E: Package 'gconf2' has no installation candidate
E: Package 'libgconf-2-4' has no installation candidate
```

fili, 2024-09-03:
Removing these packages from DEBIAN/control depends line seems to solve the problem (similar to [this](https://github.com/Eugeny/tabby/issues/9457)). gconf is [deprecated](https://packages.debian.org/bullseye/gconf2), and Gnome does not use it anymore.

littledog33, 2024-09-03:
It took some scooting around, but a couple of us figured out that one could use the balena etcher Linux zip file to successfully install the program. Download the zip file to the folder of your choice, and then extract it there. There is an ownership and permissions issue with the file chrome-sandbox. So run 
  
sudo chown root /path you downloaded to/balena-Etcher-linux-x64/chrome-sandbox 
sudo chmod 4755 /pathyoudownloaded to/balena-Etcher-linux-x64/chrome-sandbox 
/path you downloaded to/balena-Etcher-linux-x64/balena-etcher 
  
When I did this, the software installed and ran.

fili, 2024-09-03:
If you go with the updating the `.deb` file option:

Extract the `.deb` file, e.g. `balena-etcher_1.19.22_amd64.deb`, using the default decompress option in Nautilus.

Next, go into the folder, balena-etcher_1.19.22_amd64/DEBIAN and remove the following dependencies from the `control` file (you can use any editor for that):

```
gconf2
gconf-service
libgconf-2-4
```

Next repackage the folder as an installable `.deb` file again:

```
dpkg-deb --build balena-etcher_1.19.22_amd64/
```

And install it using:

```
sudo dpkg -i balena-etcher_1.19.22_amd64.deb
```

As @littledog33 mentioned, I also experienced chrome sandbox issues which I solved in this way:

```
sudo mkdir -p /opt/balenaEtcher/
sudo ln -s /usr/lib/balena-etcher/chrome-sandbox /opt/balenaEtcher/chrome-sandbox
sudo chown root:root /usr/lib/balena-etcher/chrome-sandbox
sudo chmod 4755 /usr/lib/balena-etcher/chrome-sandbox
```

Verify with
```
ls -l /usr/lib/balena-etcher/chrome-sandbox
```

I opted for the symlink option as I got an warning where it was unable to find `/opt/balenaEtcher/chrome-sandbox`.

luzan8, 2024-09-05:
Made what @fili suggested, but still getting some errors at application start. I don't know if it works fine, but installer worked and now app is starting.

alekhinexxx, 2024-09-06:
thanks you @fili its work !!

Scherer81, 2024-09-24:
> 
> Extract the `.deb` file, e.g. `balena-etcher_1.19.22_amd64.deb`, using the default decompress option in Nautilus.
> 

You could extract deb contents using following command.

```
dpkg-deb --raw-extract balena-etcher_1.19.21_amd64.deb balena-etcher_1.19.21_amd64/
```

It creates all structure needed to rebuild it after doing modifications pointed. 

In my case I was able to install and use balena just removing broken dependencies and rebuilding deb package. 

Great tips @fili and @littledog33 !

Griff-au, 2024-10-04:
Many thanks @fili.  Worked a treat.
The only other thing was that the install complained about `libgdk-pixbuf2.0-0`.  I installed with :
`sudo apt install libgdk-pixbuf2.0-0`.