balena-io/etcher #4309: E: Unable to locate package balena-etcher_1.19.21_amd64.deb

paranjay-patel, 2024-09-03:
- **Etcher version:**  1.19.21
- **Operating system and architecture:** Ubuntu 24.04.1 LTS x86_64
- **Image flashed:** N/A
- **What do you think should have happened:** `sudo apt install balena-etcher_1.19.21_amd64.deb` should have install the package
- **What happened:**
```
sudo apt install balena-etcher_1.19.21_amd64.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package balena-etcher_1.19.21_amd64.deb

```
- **Do you see any meaningful error information in the DevTools?** 
N/A

TheSuinoNick, 2024-09-08:
try: 
sudo apt install ./balena-etcher_1.19.21_amd64.deb

paranjay-patel, 2024-09-09:
```
sudo apt install ./balena-etcher_1.19.21_amd64.deb
[sudo] password for paranjay: 
Reading package lists... Done
E: Unsupported file ./balena-etcher_1.19.21_amd64.deb given on commandline
```

dm732, 2024-09-15:
I have this issue as well.

lurch, 2025-02-22:
See https://github.com/balena-io/etcher/issues/4408#issuecomment-2676215608 for the solution to "Unsupported file"