balena-io/etcher #4312: It dosent let me select my external hard drive :(

UzoroZ, 2024-09-07:
- **Etcher version:1.19.21** 
- **Operating system and architecture:Windows 11** 
- **Image flashed:Linuix mint iso file** 
- **What do you think should have happened:to let me select my external hard drive** 
- **What happened:it dosent let me use my external hard drive (cant select it, like it dosent let me)**
- **Do you see any meaningful error information in the DevTools? im bad in english sorry i cant understand this question**

GitKqt, 2024-10-01:
same issue on same OS (Windows 11 Home, Version 10.0.22000 Build 22000). After selecting the file I cannot click "select target", I am also unable to select "flash from file" again. 
[image]

GitKqt, 2024-10-01:
the issue also occurs when running as administrator

GitKqt, 2024-10-01:
currently as a workaround I'm using USBimager
https://bztsrc.gitlab.io/usbimager/