balena-io/etcher #4316: Stops at 27%

MarkRS-UK, 2024-09-13:
- **Etcher version: 
- 1.19.21 (with fix as https://github.com/balena-io/etcher/issues/4150#issuecomment-2295023596)

- **Operating system and architecture: Fedora 38 (x86_64) Live USB disk 

- **Image flashed: haos x86_64 generic 13.1

- **What do you think should have happened: Install OS image on target ssd

- **What happened:
-  Flash install goes along up to 27% abd then stops.
- The disk (112 Gb) shows several haos partitions, totally ~1.8Gb and ~4Gb unallocated!  Seems to have "lost"  ~108Gb!
- Free space on Fedora installation shows as more than 1Gb.
- I have done this three or four times now, always with exactly this result, stops at 27%

- **Do you see any meaningful error information in the DevTools?
- No error. Console shows installation up to 27% without error then just stops.

aceoffice, 2024-09-14:
I also face exactly the same problem!! stuck at 27%.. That's strange. Any experts out there to help?

MarkRS-UK, 2024-09-14:
Thank you @aceoffice , that worked perfectly!
Now you mention it I see the HA installation guide does say to try that :(
I don't s'pose there's a fix for that, so I'll close this issue.

closed 2024-09-14 by MarkRS-UK