balena-io/etcher #4319: not etching to usb drive

cooper6869, 2024-09-18:
- 1.19.21
- windows 10
- ubuntu-24.04-desktop-amd64
- drive should of been etched
- drive did not etch
- posabley
-   ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.19.21, Type = local
index.js:140 Spawning unpriviledged sidecar on port 3434
3index.js:2 Starting...
index.js:2 C:\Users\cooper.tadin.00\AppData\Local\balena_etcher\app-1.19.21\resources\etcher-util.exe
index.js:2 etcher-irhnyc
index.js:2 Retrying to connect to child process in 1000... 1 / 10
index.js:2 etcher-irhnyc
index.js:2 Retrying to connect to child process in 1000... 2 / 10
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 5 events"}
(anonymous) @ index.js:2
index.js:2 etcher-irhnyc
index.js:2 Retrying to connect to child process in 1000... 3 / 10
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 5 events"}
(anonymous) @ index.js:2
index.js:2 etcher-irhnyc
index.js:2 Retrying to connect to child process in 1000... 4 / 10
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 etcher-irhnyc
index.js:2 Retrying to connect to child process in 1000... 5 / 10
index.js:2 etcher-irhnyc
index.js:2 Retrying to connect to child process in 1000... 6 / 10
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 6 events"}
(anonymous) @ index.js:2
index.js:2 etcher-irhnyc
index.js:2 Retrying to connect to child process in 1000... 7 / 10
index.js:2 etcher-irhnyc
index.js:2 CHILD READY
index.js:2 start heartbeat
index.js:2 CHILD LOG: Ready ...
index.js:2 CHILD LOG: Scan requested
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 5 events"}
(anonymous) @ index.js:2
index.js:2 CHILD LOG: sourceMetadata requested
2index.js:2 Starting...
2index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 6 events"}
(anonymous) @ index.js:2
2index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 2 events"}
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count
(anonymous) @ index.js:2
2index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count
(anonymous) @ index.js:2
7index.js:2 Starting...
index.js:2 Spawning priviledged sidecar on port 3435
index.js:2 C:\Users\cooper.tadin.00\AppData\Local\balena_etcher\app-1.19.21\resources\etcher-util.exe
index.js:2 ... with privileges ...
index.js:2  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices
index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout'
index.js:2 Starting...
index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout'
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout'
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout'
index.js:2 Starting...
2index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout'
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
3index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout'
index.js:2 Starting...
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
2index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout'
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
2index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout'
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout'
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count
(anonymous) @ index.js:2
index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout'
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
4index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout'
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
4index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout'
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count
(anonymous) @ index.js:2
7index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout'
index.js:2 Error spawning child process Error: Elevation timeout
    at index.js:355:7719
    at r (index.js:2:329469)
(anonymous) @ index.js:2
index.js:2 Starting...
index.js:2 Error: Error spawning the child process
    at t.spawnChildAndConnect (index.js:140:44239)
    at async f (index.js:140:45207)
    at async t.flash (index.js:140:47000)
    at async S (index.js:140:51638)
    at async E.tryFlash (index.js:140:53449)
(anonymous) @ index.js:2
2index.js:2 Starting...
media/icon.png:1 
        
        
       Failed to load resource: net::ERR_FILE_NOT_FOUND
5index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count
(anonymous) @ index.js:2
index.js:2 Starting...
5index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count
(anonymous) @ index.js:2
index.js:2 Starting...
5index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count
(anonymous) @ index.js:2

yavi337, 2024-09-19:
**try to use old version like balenaEtcher-Portable-1.18.9**

yavi337, 2024-09-19:
> **try to use old version like balenaEtcher-Portable-1.18.9**

> * 1.19.21
> * windows 10
> * ubuntu-24.04-desktop-amd64
> * drive should of been etched
> * drive did not etch
> * posabley
> * _**| |     | |
>   | |** | |_ _**| |**   ___ _ __
>   |  **|| __/ __| '_ \ / _ \ '__|
>   | |_**| || (__| | | |  **/ |
>   _**_/ _____|_| |_|___|_|
> 
> Interested in joining the Etcher team? Drop us a line at [join+etcher@balena.io](mailto:join+etcher@balena.io)
> 
> Version = 1.19.21, Type = local index.js:140 Spawning unpriviledged sidecar on port 3434 3index.js:2 Starting... index.js:2 C:\Users\cooper.tadin.00\AppData\Local\balena_etcher\app-1.19.21\resources\etcher-util.exe index.js:2 etcher-irhnyc index.js:2 Retrying to connect to child process in 1000... 1 / 10 index.js:2 etcher-irhnyc index.js:2 Retrying to connect to child process in 1000... 2 / 10 index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 5 events"} (anonymous) @ index.js:2 index.js:2 etcher-irhnyc index.js:2 Retrying to connect to child process in 1000... 3 / 10 index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 5 events"} (anonymous) @ index.js:2 index.js:2 etcher-irhnyc index.js:2 Retrying to connect to child process in 1000... 4 / 10 index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"} (anonymous) @ index.js:2 index.js:2 etcher-irhnyc index.js:2 Retrying to connect to child process in 1000... 5 / 10 index.js:2 etcher-irhnyc index.js:2 Retrying to connect to child process in 1000... 6 / 10 index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 6 events"} (anonymous) @ index.js:2 index.js:2 etcher-irhnyc index.js:2 Retrying to connect to child process in 1000... 7 / 10 index.js:2 etcher-irhnyc index.js:2 CHILD READY index.js:2 start heartbeat index.js:2 CHILD LOG: Ready ... index.js:2 CHILD LOG: Scan requested index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"} (anonymous) @ index.js:2 index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 5 events"} (anonymous) @ index.js:2 index.js:2 CHILD LOG: sourceMetadata requested 2index.js:2 Starting... 2index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"} (anonymous) @ index.js:2 index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 6 events"} (anonymous) @ index.js:2 2index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count (anonymous) @ index.js:2 index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 2 events"} (anonymous) @ index.js:2 index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count (anonymous) @ index.js:2 2index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"} (anonymous) @ index.js:2 index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count (anonymous) @ index.js:2 7index.js:2 Starting... index.js:2 Spawning priviledged sidecar on port 3435 index.js:2 C:\Users\cooper.tadin.00\AppData\Local\balena_etcher\app-1.19.21\resources\etcher-util.exe index.js:2 ... with privileges ... index.js:2 0 devices, 0% at 0.00 MB/s (total 0.00 MB/s) with 0 failed devices index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout' index.js:2 Starting... index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout' index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"} (anonymous) @ index.js:2 index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout' index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"} (anonymous) @ index.js:2 index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout' index.js:2 Starting... 2index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout' index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"} (anonymous) @ index.js:2 3index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout' index.js:2 Starting... index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"} (anonymous) @ index.js:2 2index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\Users\COOPER~1.00\AppData\Local\Temp\8b642da8-65bf-49af-92e9-7bef155a2399\stdout' index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"} (anonymous) @ index.js:2 2index.js:2 Error while reading flasher elevation script output Error: ENOENT: no such file or directory, open 'C:\U [... cut ...] rn]: {code: 0, error: "Status 'unknown' provided for 1 events"} (anonymous) @ index.js:2 index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count (anonymous) @ index.js:2 index.js:2 Starting... 5index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"} (anonymous) @ index.js:2 index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count (anonymous) @ index.js:2

try to use old version like balenaEtcher-Portable-1.18.9

cooper6869, 2024-09-20:
ok thanks @yavi337