balena-io/etcher #4324: index.js:2 TypeError: (0 , h.requestMetadata) is not a function

sctech-tr, 2024-09-22:
- **Etcher version:** 1.19.21 installed via RPM package
- **Operating system and architecture:** Fedora Linux 40 x64 (KDE Plasma)
- **Image flashed:** Slax Linux's latest version, i modified the bootscreen and packed it again. it still fails at the normal iso too
- **What do you think should have happened:** i wanted it to etch my iso
- **What happened:** it gave this error about non-functional functions (get it?)
[image]
- **Do you see any meaningful error information in the DevTools?** 
this:  
index.js:2 TypeError: (0 , h.requestMetadata) is not a function
at index.js:140:1512
at async W.openImageSelector (index.js:140:2626)

sivendsg, 2024-09-23:
i am having the same problom adn i dont know how to fix it someone help

sctech-tr, 2024-09-23:
use something like unetbootin or rufus until it gets fixed
https://rufus.ie
https://unetbootin.github.io

lurch, 2024-09-23:
Probably a duplicate of #4150 ?

SamuelBanya, 2024-09-28:
Everyone, use an older version of the AppImage release like Version 1.13.0 found on this release page:
- https://github.com/balena-io/etcher/releases?page=5

sctech-tr, 2024-09-28:
Duplicate of #4150

rglidden, 2024-10-10:
@sctech-tr A fix for the .rpm package was merged into 1.19.24 today and is working on my Fedora 40 install. Can you give that version a try? https://github.com/balena-io/etcher/releases/tag/v1.19.24