balena-io/etcher #4325: Fix the problem with NVME M2 SSDs

Neltherion, 2024-09-24:
Flashing an image with Balena on an NVME SSD and then using it on a Jetson device would result in an error described [here](https://forums.developer.nvidia.com/t/jetson-xavier-nx-devkit-cannot-be-flashed-on-nvme/280796/1) and the fix is also in [this post](https://forums.developer.nvidia.com/t/jetson-xavier-nx-devkit-cannot-be-flashed-on-nvme/280796/8).

Is this a problem that should be fixed by the Balena team or NVIDIA?

acostach, 2024-09-24:
Hi @Neltherion , can you please share what exact device type and what balenaOS version are you using?

Neltherion, 2024-09-24:
Device => Jetson Orin Nano
Balena Etcher => 1.19.21

The Jetson Orin Nano was flashed with NVIDIA SDK Manager successfully first but before or even after the flash, I could not boot up the Jetson with the `jp60-orin-nano-sd-card-image.zip` etched on an NVME M.2 SSD.

The errors are exactly the same as was described [here](https://forums.developer.nvidia.com/t/jetson-xavier-nx-devkit-cannot-be-flashed-on-nvme/280796/1) and the fix was exactly the same described [here](https://forums.developer.nvidia.com/t/jetson-xavier-nx-devkit-cannot-be-flashed-on-nvme/280796/8).

acostach, 2024-09-24:
Thanks for clarifying @Neltherion , so the file you flashed on the NVME with Etcher is the Nvidia Ubuntu image, and to get it to work you had to modify extlinux.conf in it and change the root in the kernel cmdline to be the first partition of the nvme, instead of the first partition of the sd-card. This is expected behavior because the Ubuntu sd-card image contents are designed to be flashed to an sd-card and not to a NVME. Etcher writes the image as is, without modifying the contents of extlinux.conf or other files.

Neltherion, 2024-09-24:
Thanks for the clarification.

closed 2024-09-24 by Neltherion