balena-io/etcher #4327: Receiving 'Error: (0, h.requestMetadata) is not a function' For Latest AppImage Version Of Balena Etcher On Fedora Workstation Linux (x64)

SamuelBanya, 2024-09-28:
- **Etcher version:** 
- Version 1.19.21-x64.AppImage
- **Operating system and architecture:** 
- Linux, x64
- **Image flashed:** 
- mr-fusion.img
- **What do you think should have happened:** 
- I expected Balena Etcher to be able to use an .img file to burn to a USB flash drive
- **What happened:**
- When I attempt to load the .img file on the first 'Flash from file' step, I receive this error:
```
Error opening source

Something went wrong while opening /home/sam/Downloads/mr-fusion-v2.9.img/mr-fusion.img

Error: (0, h.requestMetadata) is not a function
```
- **Do you see any meaningful error information in the DevTools?** 
- Not applicable as I am running the App Image version of this on Linux, specifically Fedora Workstation (latest as of September 2024).

My solution:
- I had to resort to an older version of Balena Etcher, specifically Version 1.13.0, which can be found here:
- https://github.com/balena-io/etcher/releases?page=5

I mention this here so someone later can give me props for this, since thankfully I am a bit used to this as it always feels like the latest version of Balena Etcher never works nicely on Linux, and I always have to go back to some older version for basic functionality to work.

Anyway, hoping this helps someone else out, and maybe a note to QA these releases for Linux before putting it out there, a bit frustrating but understanding I guess.

Fabxx, 2024-09-28:
Can confirm on Debian 12 as well

adriansev, 2024-10-02:
Also on Alma9 .. trying to write an Alma9 dvd iso and some other image .. factually at this moment balena-etcher is not usable/is broken

jeffiscow2, 2024-10-02:
yeah I had to install 1.13.0 .rpm to get it to work on fedora 40

lurch, 2024-10-02:
See also #4150

Spacemanit1709, 2024-10-04:
issue also on windows, but, I downgraded to 1.18 and it worked.

DarthRidar, 2024-11-01:
> Problem anfenstern, aber stuftee auf 1,18 und es die herabsen.
Thanks. That works.
v1.19.25 is broken in Arch with AppImage.

DarthRidar, 2024-11-01:
> Problem anfenstern, aber stufteee auf 1,18 und es die herabsen.

Thanks. That works.
v1.19.25 is broken in Arch with AppImage.

mlizzi, 2024-11-02:
Same for 2017 Mac

Onaolapo-11, 2024-11-16:
I'm trying to install Ubuntu 24.04 LTS on hp elitebook 8540p and had this error. But I'd try this older version of BalenaEtcher. Thank you.

369px, 2024-11-21:
I had this same problem but fixed it! I think I know why it's happening.
Since Apple now doesn't recognize etcher as "unknown source" anymore, it opens the app WITHOUT admin permissions!
This is something really rare on apple devices so we're used to not being able to open an app unless we click open by pressing the option button simultaneously (so it gets admin permissions).

I closed etcher, opened it again with the option button and I flashed  a bootable image in my sd card correctly!
Hope this helps!

jikuolu, 2024-11-26:
> I had this same problem but fixed it! I think I know why it's happening. Since Apple now doesn't recognize etcher as "unknown source" anymore, it opens the app WITHOUT admin permissions! This is [...]

I had the same problem and worked following your solution! thanks