balena-io/etcher #4328: Not able to install etcher on ubuntu 24.10

agrawal-pratham, 2024-09-29:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **What do you think should have happened:** 
- **What happened:**
- **Do you see any meaningful error information in the DevTools?** 

[image]
[image]
[image]

ftmiranda, 2024-09-29:
Same thing happens on Ubuntu 24.04.1
Did anyone notice that?

TheRiseOfDavid, 2024-10-01:
I have the same problem 
* Depends: gconf-service but it is not installable
* Depends: gconf2 but it is not installable
* Depends: libgconf-2-4 but it is not installable

I attention the issue since [Problems with installation due to gconf-service and gconf2 deprecated in ubuntu 23.10](https://github.com/Eugeny/tabby/discussions/9456)

Please help me to solve it. 
[image]

lurch, 2024-10-01:
https://github.com/balena-io/etcher/issues?q=is%3Aissue+is%3Aopen+gconf2

agrawal-pratham, 2024-10-03:
@lurch @ftmiranda @TheRiseOfDavid @otavio 
I solved it by simply installing using the zip file method.

closed 2024-10-03 by agrawal-pratham

TheRiseOfDavid, 2024-10-03:
@agrawal-pratham Thank you for this solution