balena-io/etcher #4330: Echer 1.19.21 "Something went wrong. If it is a compressed image, please check that the archive is not corrupted".

saulpugg, 2024-10-06:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **What do you think should have happened:** 
- **What happened:**
- **Do you see any meaningful error information in the DevTools?**

saulpugg, 2024-10-06:
The software doesn't allow me to flash the USB stick. It stops showing the message shown on the title.

gramtech, 2024-10-11:
Hey, I am having the same issue, I have tried a couple different USB sticks, one of which I have used before with Etcher

    Etcher version: 1.19.25
    Operating system and architecture: MacOS Somona, Intel 
    Image flashed: ubuntu-mate-24.04.1-desktop-amd64.iso 
    What happened: Received the following error:

Something went wrong. If it is a compressed image, please check that the archive is not corrupted.
The writer process ended unexpectedly

SeQwintial, 2024-10-11:
[image]
I have also received the same error as the above comment; in addition, 4 separate windows popped up during the validation process telling me I needed to format the new drives (each one is partitioned from the flash, i'm assuming) before I can use them. Is Windows getting in the way of the final validation?

juniorr001, 2024-10-11:
I solved it by running as administrator.

SeQwintial, 2024-10-11:
> I solved it by running as administrator. right click on the shortcut > run as administrator.

I have unfortunately tried this, as well as disabling my antivirus/windows defender, and trying it in a different usb slot. Attempting to boot the kobo with the sd card in it also does nothing (in the case that the validation error was irrelevant and the flashing had worked properly)

2ZZ, 2024-11-11:
This is happening every time on MacOS for me, I tried 3 different Macbooks, different USB sticks and all on the latest version of Etcher.

fatso83, 2024-11-17:
@saulpugg There is absolutely nothing to go buy in your bug reports. It's near useless when you don't even bother filling out the template. Like which version, which os, ... how do you expect anyone to be able to work with this?

2ZZ, 2024-11-17:
> I first tried flashing the file in the file.img.xz format, but it failed. Then I tried extracting it to somefile.img first and then flashing it, and that worked, all on MacOS.

My failures were with .xz as well.

SeQwintial, 2024-11-18:
> > I first tried flashing the file in the file.img.xz format, but it failed. Then I tried extracting it to somefile.img first and then flashing it, and that worked, all on MacOS.
> 
> My failures were with .xz as well.

I've solved my problem with this, and it turns out I had to extract the .xz file _first_ before attempting to flash it. Worked on first attempt, Win10.

[image]
(Note the "0-N905C file" extension - this was the one that worked)

MrTechGadget, 2024-11-20:
I am also experiencing this issue on macOS with 1.19.25 on a .img.xz image that I have used dozens of times with no issues with 1.7.8. Apparently there has been a feature regression.

4b11b4, 2024-11-27:
Also having this issue

macOS 14.0, etcher 1.19.21

I have an .xz image
(from Luxonis' [RPI_OS](https://drive.google.com/file/d/1TdnvD71W7avGqHO0fQILSiZGDYUeE-OV/view?usp=drive_link))
that I decompressed and added a .img extension to
It seemed that selecting the .xz would result in nothing happening in the interface (couldn't proceed to selecting target)

Attempts:
1. Failure
From dev tools console, it seems I got to 92% then there was an unexpected write failure.
2. Failure
Verifying 1 device, 100% at 19.90 MB/s (total 19.90 MB/s) eta in 0s with 0 failed devices
index.js:2 fail event
index.js:2 {size: 31266438656, isVirtual: false, enumerator: 'DiskArbitration', logicalBlockSize: 512, raw: '/dev/rdisk4', …}blockSize: 512busType: "USB"busVersion: nulldescription: "Mass Storage Device Media"device: "/dev/disk4"devicePath: "IODeviceTree:/arm-io/usb-drd0@2280000/usb-drd0-port-hs@00100000"displayName: "/dev/disk4"enumerator: "DiskArbitration"error: nullisCard: falseisReadOnly: falseisRemovable: trueisSCSI: falseisSystem: falseisUAS: nullisUSB: trueisVirtual: falselogicalBlockSize: 512mountpoints: [{…}]partitionTableType: "mbr"raw: "/dev/rdisk4"size: 31266438656[[Prototype]]: Object
index.js:2 {name: 'Error', message: 'Unmount failed, invalid drive', stack: 'Error: Unmount failed, invalid drive'}message: "Unmount failed, invalid drive"name: "Error"stack: "Error: Unmount failed, invalid drive"[[Prototype]]: Object
index.js:2 Starting...
index.js:2 Starting...
index.js:2 Flash results {}
index.js:2 {}
index.js:2 CHILD: flash done {results: {…}}results: bytesWritten: 7624622592devices: {failed: 1, successful: 0}errors: [Error: Unmount failed, invalid drive]sourceMetadata: {size: 8787493376, name: 'RPi_64bit_OS.img', blocks: Array(13), blockmappedSize: 7624622592}[[Prototype]]: Object[[Prototype]]: Object
index.js:2 Flash results {results: {…}}results: bytesWritten: 7624622592devices: {failed: 1, successful: 0}errors: [Error: Unmount failed, invalid drive]sourceMetadata: {size: 8787493376, name: 'RPi_64bit_OS.img', blocks: Array(13), blockmappedSize: 7624622592}[[Prototype]]: Object[[Prototype]]: Object
index.js:2 Starting...
index.js:2 Error: The writer process ended unexpectedly
    at i (index.js:353:5238)
    at t.createUserError (index.js:353:6464)
    at m (index.js:140:45502)
    at index.js:140:46208
    at L.<anonymous> (index.js:140:42787)
    at L.emit (node:events:518:28)
    at e.exports.H (index.js:2:2317562)
    at e.exports.emit (node:events:518:28)
    at index.js:2:2293382
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)

4b11b4, 2024-11-27:
one note is that I still get errors after success in the console (v1.91.25)

`Verifying 1 device, 100% at 20.00 MB/s (total 20.00 MB/s) eta in 0s with 0 failed devices
index.js:2 CHILD: flash done {results: {…}}results: averageFlashingSpeed: 11454126.53192442bytesWritten: 7624622592devices: {failed: 0, successful: 1}errors: []sourceMetadata: {size: 8787493376, name: 'RPi_64bit_OS.img', blocks: Array(13), blockmappedSize: 7624622592}[[Prototype]]: Object[[Prototype]]: Object
index.js:2 Flash results {results: {…}}cancelled: falseresults: averageFlashingSpeed: 11454126.53192442bytesWritten: 7624622592devices: {failed: 0, successful: 1}errors: []sourceMetadata: {size: 8787493376, name: 'RPi_64bit_OS.img', blocks: Array(13), blockmappedSize: 7624622592}[[Prototype]]: Objectskip: false[[Prototype]]: Object
index.js:2 got results {results: {…}}cancelled: falseresults: averageFlashingSpeed: 11454126.53192442bytesWritten: 7624622592devices: {failed: 0, successful: 1}errors: []sourceMetadata: {size: 8787493376, name: 'RPi_64bit_OS.img', blocks: Array(13), blockmappedSize: 7624622592}[[Prototype]]: Objectskip: false[[Prototype]]: Object
index.js:2 Starting...
index.js:2 removed flashing flag
media/icon.png:1 
        
        
       GET file:///Applications/balenaEtcher.app/Contents/Resources/app.asar/.webpack/renderer/main_window/media/icon.png net::ERR_FILE_NOT_FOUND
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 2 events"}
(anonymous) @ index.js:2
e.warn @ index.js:2
e.handleResponse @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
a @ index.js:355
Promise.then (async)
l @ index.js:355
(anonymous) @ index.js:355
h @ index.js:355
e.send @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
(anonymous) @ index.js:355
h @ index.js:355
e.flush @ index.js:2
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
e.addToQueue @ index.js:2
(anonymous) @ index.js:2
e.execute @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
a @ index.js:355
Promise.then (async)
l @ index.js:355
a @ index.js:355
Promise.then (async)
l @ index.js:355
a @ index.js:355
Promise.then (async)
l @ index.js:355
a @ index.js:355
Promise.then (async)
l @ index.js:355
a @ index.js:355
Promise.then (async)
l @ index.js:355
(anonymous) @ index.js:355
h @ index.js:355
e.apply @ index.js:2
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.scheduleApply @ index.js:2
(anonymous) @ index.js:2
e.push @ index.js:2
[... 528 lines cut ...]
e.process @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
(anonymous) @ index.js:355
h @ index.js:355
t.process @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
(anonymous) @ index.js:355
h @ index.js:355
e.dispatch @ index.js:2
e.track @ index.js:2
(anonymous) @ index.js:2
e.track @ index.js:2
(anonymous) @ index.js:140
t.logEvent @ index.js:140
processTicksAndRejections @ node:internal/process/task_queues:95
await in processTicksAndRejections (async)
t.flash @ index.js:140
processTicksAndRejections @ node:internal/process/task_queues:95
await in processTicksAndRejections (async)
processTicksAndRejections @ node:internal/process/task_queues:95
index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count`

YorkshireHillbilly, 2024-12-18:
> I've solved my problem with this, and it turns out I had to extract the .xz file _first_ before attempting to flash it. Worked on first attempt, Win10.

This saved me from as complete temper tantrum! I'd been trying to fix Home Assistant all night and deeded to reinstall it on my NUC but I couldn't get the blasted image t write properly.
Thanks @SeQwintial

37ozkanbingol, 2026-01-06:
> v1.18.11 worked for me

sorry dont worked for my macos . are you help to me. my macos version catalina 10.15.7.   thankyou

JerRoyDev, 2026-02-03:
I had the same issue, but then tried write image with "Raspberry Pi Imager" instead of BalenaEtcher.  It worked yo :)

jbwharris, 2026-09-19:
Experienced this with Tahoe 26.6. Was resolved by using rpi-imager instead.

random-donuts, 2026-11-01:
Same issue with MacOS 27.0.1
Attempted to start with `sudo` from Terminal. Doesn't show the error UI as above, but the progress bar never moves and Terminal shows:

`sandbox_extension_issue_file failed for /Applications/balenaEtcher.app/Contents/Frameworks/balenaEtcher Helper.app/Contents/Resources: 1 (Operation not permitted)`

The app never request for any password if it was launch under Sudo

Command I ran:
`sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher`

Output:
```
(electron) 'session.getPreloads' is deprecated and will be removed. Please use 'session.getPreloadScripts' instead.
(electron) 'session.setPreloads' is deprecated and will be removed. Please use 'session.registerPreloadScript' instead.
Language changed to: en
ready-to-show: 321.724ms
packageUpdatable false
sandbox_extension_issue_file failed for /Applications/balenaEtcher.app/Contents/Frameworks/balenaEtcher Helper.app/Contents/Resources: 1 (Operation not permitted)
```