balena-io/etcher #4331: my Ubuntu isn't installing Balena Etcher

yahuda-sudo, 2024-10-07:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **What do you think should have happened:** 
- **What happened:**
- **Do you see any meaningful error information in the DevTools?**

Peter-MJ-Parker, 2024-10-09:
I got this same thing. I am on Ubuntu 24.04.1. Here's my workaround.
- Add Ubuntu 22.04 sources to system
```sudo nano /etc/apt/sources.list.d/jammy-sources.list```
In this file input the following:
```
deb http://archive.ubuntu.com/ubuntu/ jammy main restricted universe multiverse
deb http://archive.ubuntu.com/ubuntu/ jammy-updates main restricted universe multiverse
deb http://archive.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
```
Save with CTRL + X, Enter, Y

- Update packages
```sudo apt update```

- Install the three missing packages
```sudo apt install gconf-service gconf2 libgconf-2-4```

- Make sure root has direct access to the folder and file where you saved the .deb of balena-etcher
```sudo chown root:root /path/to/balena-etcher_1.19.21_amd64.deb
sudo chmod 644 /path/to/balena-etcher_1.19.21_amd64.deb
```
- Install balena-etcher
  - Open the folder where the .deb file is located
  - Right click inside the folder and Open in Terminal
  ```sudo apt install ./balen-etcher_1.19.21_amd64.deb```
  
Hope this works for you!

lurch, 2024-10-09:
This problem has already been reported a few times before https://github.com/balena-io/etcher/issues?q=is%3Aissue+is%3Aopen+gconf2