balena-io/etcher #4332: Kubuntu 24.04.1 LTS x86_64 - Etcher won't run.

twylightproto, 2024-10-07:
- **Etcher version:** Latest
- **Operating system and architecture:** Kubuntu 24.04.1 LTS 64-bit
- **Image flashed:** N/A
- **What do you think should have happened:**  The app should've launched.
- **What happened:**Well, it didn't
- **Do you see any meaningful error information in the DevTools?** Wouldn't, it just doesn't do anything upon 2x click

When trying to launch the latest Etcher, it just does nothing (64-bit, so no appimage, an it's an executeable with other shet). It IS set to 'Is Executable'. So, trying to run in Konsole, says that it doesn't see the file.

twylightproto, 2024-10-07:
Running in konsole just got "[11356:1007/103641.321436:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /home/identitycrisis/Documents/balenaEtcher-linux-x64/chrome-sandbox is owned by root and has mode 4755.
"

lurch, 2024-10-09:
Related to #4314 ?

a440guy, 2024-10-18:
cd <location of extracted fiiles>
sudo chown root ./chrome-sandbox
sudo chmod 4755 chrome-sandbox

HernandoR, 2025-01-06:
I could not recall the exact URL that made me I finally solve this problem. I guess it was a Cursor (the IDE) issue or an Obsidian discussion.

The reason why some (or all, in my case) AppImages cannot start without `--no-sandbox` is due to a change of default configuration appended in Kubuntu 24.04 (yeah, so exact). You could configure it with the following:
``` 
sudo sysctl -w kernel.apparmor_restrict_unprivileged_userns=0
```