balena-io/etcher #4338: etcher destroyed my SANDISK USB drives

BestNoob, 2024-10-15:
1x SanDisk Ultra USB 3.0
1x SanDisk Ultra Luxe USB 3.2
1x SanDisk Ultra Flair USB 3.0

All 3 drives are defective AFTER using belena etcher.

1x is undetectable
1x is locked status
1x is not formatable

MarkyMarkDE, 2024-10-17:
Which etcher Version and on which OS?
These are minimalist information for creating a new Bug report

Samzeu2u, 2024-10-22:
same happened with me, 3 Sandisk usb drives. but i used rufus to flash on these otherwise obsolete drives and it worked. etcher wasnt letting me flash on these drives.

ChristopherTheBoss, 2024-11-14:
this software was recommended by a YouTube channel
it has destroyed a 256 GB Samsung flash drive of mine
it is malware. etcher indeed.

lurch, 2024-11-14:
https://blog.balena.io/did-etcher-break-my-usb-sd-card/