balena-io/etcher #4343: etcher doesn't flash

Duck-Dude, 2024-10-26:
- **Etcher version: latest** 
- **Operating system and architecture: Kubuntu 24.10 x64 ** 
- **Image flashed: the kubuntu 24.10 install iso** 
- **What do you think should have happened: I have no idea**
- **What happened: it won't let me flash my usb drive with any image saying (Error spawning the child process.)**
- **Do you see any meaningful error information in the DevTools? no**
[image]

chrisbadley, 2024-10-27:
Exact same issue. Using windows 10 trying to flash proxmox 8.2-2

Scotiangold, 2024-10-29:
Same issue. Same error message. Downloaded Linuxlite 7.2 iso. Same error message trying to create usb flash drive on Windows 11 desktop computer and Balena Etcher.

chrisbadley, 2024-10-29:
> Same issue. Same error message. Downloaded Linuxlite 7.2 iso. Same error message trying to create usb flash drive on Windows 11 desktop computer and Balena Etcher.

I overcame this problem by simply running etcher as admin

fengqianliuli, 2024-10-31:
> > 同样的问题。相同的错误消息。下载 Linuxlite 7.2 iso。尝试在 Windows 11 台式计算机和 Balena Etcher 上创建 USB 闪存驱动器时出现相同的错误消息。
> 
> 我通过简单地以管理员身份运行 etcher 来克服了这个问题

great！！！！！