balena-io/etcher #4349: Can't open etcher on latest MacOS

aethernet, 2024-11-07:
Hello, 

With the new restriction in the latest Macos, it's possible your existing version of Etcher won't run anymore.
The latest version available on GitHub (and https://etcher.balena.io) is signed with a new certificate that should work fine.

fabiocorneti, 2024-11-16:
Hi there,
I opened Etcher 1.29.21, installed with brew some time ago on Sequoia 15.1 and it was binned automatically by gatekeeper as "malware".

To double check it was not some malicious release, I downloaded manually 1.19.21 from the GitHub releases page, checked the SHA, and indeed it is still flagged as "malware" as the certificate was revoked:

[image]

```
spctl --assess --verbose=4 ~/Downloads/balenaEtcher.app
balenaEtcher.app: CSSMERR_TP_CERT_REVOKED
```

---

For peace of mind, can you confirm that the old certificate has been explicitly revoked upon issuing the new one?

cgruver, 2024-11-19:
Same thing happened to me with an older release.  Version 1.19.25 appears to work fine on the new MacOS release.

moreati, 2024-11-20:
Quoting the error messages, to help people searching the Internet find this

> The application “balenaEtcher.app” can’t be opened.
> Malware Blocked and Moved to Bin
> “balenaEtcher.app” was not opened because it contains malware. This action did not harm your Mac."

This was balenaEtcher 1.18.11 on macOS 15.1.

aethernet, 2024-11-27:
> For peace of mind, can you confirm that the old certificate has been explicitly revoked upon issuing the new one?

Yes it's been revoked.

The Macos messaging is unfortunate as it's not malware, but a certificate revocation.

OptimusGREEN, 2024-12-01:
latest version seems ok.

z3cka, 2025-01-02:
[image]

updated via home brew to `1.19.25` and a still get the "“balenaEtcher.app” will damage your computer." message when running on macOS Sonoma 14.7.2

Any ideas on how to get around this? I've been using etcher on a mac for years without issue; I know it will not harm my computer.

z3cka, 2025-01-02:
[image]

lol, apparently there's a checkbox for this. Right click application -> get info -> check "Override Malware Protection" and Bob's your uncle. source: https://cybernews.com/malware/bypass-will-damage-your-computer-malware-warning/ _(this is really stupid)_

marleyjaffe, 2025-03-30:
Experiencing this issue on Balena `Version 2.1.0 (2.1.0)` as well. @z3cka's `Override Malware Protection` seems to have solved the issue.

osy, 2025-07-19:
@aethernet can you explain exactly why the certificate was revoked? Was the certificate compromised? Was it used to sign anything malicious? Did you revoke the certificate or did Apple revoke it? How can we be sure that the latest version is not compromised?

smileBeda, 2025-09-09:
Latest version `brew install balenaetcher` works just fine.
If you are in doubt, let Mac do its thing, then reinstall it.