balena-io/etcher #4352: Flash from file does not work

MattBright01, 2024-11-10:
- **Etcher version: 1.19.25** 
- **Operating system and architecture: Windows 10 Intel Celeron 2.13 Ghz 8Gb RAM ** 
- **Image flashed: haos_generic-x86-64-13.2.img** 
- **What do you think should have happened: Open the source** 
- **What happened: Error opening Source   Error: (0 , h.requestMetadata) is not a function  **
- **Do you see any meaningful error information in the DevTools?  No**

MattBright01, 2024-11-11:
Problem solved with version 1.18
Can be dowloaded for Windows here: https://etcher.fr.uptodown.com/windows/telecharger/106963802

lurch, 2024-11-11:
> * **What happened: Error opening Source   Error: (0 , h.requestMetadata) is not a function  **

See #4150 

Be careful with downloading Etcher from 3rd-party websites as apparently there are "fake" versions #3724
Official old releases of Etcher can be downloaded from https://github.com/balena-io/etcher/releases

imagine496, 2024-11-12:
Hi, i have the same problem , and now i see an answer but lurch warns about fake websites, how can i verify that the link he gives is official? or if  Matts link is ok,, heck i havent got very far with trying to get Lynux Mint, this is my second day trying to do what should only take minutes, please advise thank you
[image]

MattBright01, 2024-11-12:
Hi, I had no problem with this link. I simply found it by searching on Google. Otherwise, you can download the version you need here on Github too, as @lurch explains.

danilogco, 2024-11-27:
Still happening with the latest version. It worked with v1.18.8.

jnorre, 2024-11-28:
Happening for me too: Official **balenaEtcher-1.19.25-x64.AppImage** from github releases on Ubuntu 24.04.01 LTS.

vmario89, 2024-12-02:
same problem on TuxedoOS

sparevermicelli, 2024-12-04:
Same issue, Win10 + 1.19.25

ShoGUN1Master, 2024-12-12:
Same issue 12.12.2024 Win 11!

carlitos816, 2024-12-13:
Same issue MACOS Ventura 13.6.9, tried two different Ubuntu ISOs

OzHershy, 2025-02-16:
I have the same issue and am using the very latest version, I am trying to load Ubuntu onto a working 5 year old HP laptop running Windows 10.