balena-io/etcher #4354: Link broken in downloaded balenaEtcher zip file

mpeastend, 2024-11-17:
- **latest Etcher version from official website (link on Ubuntu.com website for Ubuntu 24.04 LTS install instructions)** 
- **Linux Mint 19.3 Cinnamon 4.4.8 Intel© Core™ i9-9900K CPU @ 3.60GHz × 8.  Attempting to upgrade a way out-of-date OS** 
- **downloaded zip file for linux 64 bit.  After extracting, and opening file folder, all attempts to run balenaetcher result in : This link cannot be used, because its target "/home/runner/work/etcher/etcher/out/balenaEtcher-linux-x64/balena-etcher" doesn't exist.  I haven't moved the file, only extracted it to Downloads and extracted it there.  All permissions are read & write** 
Please Help!

TDuffinNTU, 2024-11-22:
You can try the appimage and deb installs in the releases section but I'd suggest the latest .deb over the .appimage because there seems to be some weird issue with the appimage for me.

I think you might need an older release too for the deb if the mint version doesn't install properly

mpeastend, 2024-11-25:
Thank you TDuffinNTU, I appreciate the response and help.  I was unable to

mpeastend, 2024-11-25:
Issue resolved with a different open source tool.