balena-io/etcher #4358: arm64 The writer process ended unexpectedly

ZhitongGuo, 2024-11-30:
- **Etcher version:** 
balenaEtcher-1.19.25-arm64
- **Operating system and architecture:**  macos, arm64
- **Image flashed:** 
jetson-nano-jp461-sd-card-image: sd-blob-b01.img
- **What do you think should have happened:** 
- **What happened:**
The writer process ended unexpectedly, unable to write. At first, the writer process's progress bar can move for 2 minutes before the error pops up. After a few trials, the error shows up immediately after I started the process. I am pretty sure that the image is not corrupted and the card works fine. I downloaded the image multiple times from the website, and unzipped it before selecting the image. 
- **Do you see any meaningful error information in the DevTools?** 

index.js:2 Error: The writer process ended unexpectedly
    at i (index.js:353:5238)
    at t.createUserError (index.js:353:6464)
    at m (index.js:140:45502)
    at index.js:140:46208
    at L.<anonymous> (index.js:140:42787)
    at L.emit (node:events:518:28)
    at e.exports.H (index.js:2:2317562)
    at e.exports.emit (node:events:518:28)
    at index.js:2:2293382
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
(anonymous) @ index.js:2
2index.js:2 Starting...
media/icon.png:1 
        
        
       Failed to load resource: net::ERR_FILE_NOT_FOUND

ZhitongGuo, 2024-12-01:
resolved after upgrading macos system.. interesting

closed 2024-12-01 by ZhitongGuo

dojoca, 2024-12-20:
Still getting this error on MacOS 12.7.6. Rebooting, formatting the USB disk doesn't help at all. Eventually gave up and used my Ubuntu box to write it.

WinstonSmith73, 2025-08-12:
I just fixed this error on macOS Sequoia 15.6 by adding Balena Etcher to the list of apps with full disk access (System settings> Privacy & Security> Full Disk Access click the + button to add to list).