balena-io/etcher #4366: Error opening iso file

MPForsyth, 2024-12-20:
- **Etcher version:** Etcher/balenaEtcher-1.19.25-x64.AppImage
- **Operating system and architecture:** Ubuntu 22.04  desktop
- **Image flashed:** Downloads/ubuntu-24.04.1-desktop-amd64.iso
- **What do you think should have happened:**should open the file
- **What happened:**Error message  - see attached
[image]

- **Do you see any meaningful error information in the DevTools?**

daniel-hayim, 2024-12-20:
I'm running windows 11 and I'm trying to make a flash usb to install MATE on another computer and I get the same error.

PixsaOJ, 2024-12-21:
Same on linux

jackrataty, 2024-12-21:
Windows 10 trying to trying to flash Mint iso.  Same error

MPForsyth, 2024-12-22:
I just tried the install of a .deb package and it works but there were some error messages
....
etting up balena-etcher (1.19.25) ...
chmod: cannot access '/opt/balenaEtcher/chrome-sandbox': No such file or directo
ry
Processing triggers for desktop-file-utils (0.26-1ubuntu3) ...
Processing triggers for gnome-menus (3.36.0-1ubuntu3) ...
Processing triggers for libc-bin (2.35-0ubuntu3.8) ...
Processing triggers for mailcap (3.70+nmu1ubuntu1) ...
N: Download is performed unsandboxed as root as file '/home/mike22/Etcher/balena-etcher_1.19.25_amd64.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
.....
so the problem is with the AppImage   file.

irhdab, 2024-12-30:
running windows 11, trying to flash ubuntu iso, i also have a same error