balena-io/etcher #4371: Program incorrectly assumes current user has admin permissions

gabefair, 2024-12-24:
- **Etcher version:** 1.19.25
- **Operating system and architecture:** MacOS 14
- **Image flashed:** None

- **What do you think should have happened:** 
When the user starts the flash process, Balena Etcher prompts for password to gain elevated permissions.
However this is done without giving the user the ability to change the username of user with administrative rights.
Program is assuming that the current user has admin rights.

- **Do you see any meaningful error information in the DevTools?** No

LangdonStIves, 2025-01-19:
Commenting to let devs know others are running into this problem too. The only workaround is to switch to a user with admin rights and run the program there, but that's clunky. 99.9% of macos software that needs elevated privileges is aware of this and presents a dialogue with both user and password fields at least when run under a user that doesn't have them, so I am sure there is some standard way to do this (i.e., no need to reinvent the wheel).

lurch, 2025-01-20:
Related to #3321

matt17r, 2025-12-23:
Workaround (without using Balena Etcher) from [linked thread](https://github.com/balena-io/etcher/issues/3321#issuecomment-2506016895) (thanks for the link @lurch).

As the unprivileged user:

- open Terminal
- switch user to admin account (`su admin_account` then enter password of admin_account)
- do a block copy as per the other thread (`sudo dd if=/Users/<depends_on_you>/Downloads/<image_name>.iso of=/dev/<removable_stick_name> bs=1m conv=noerror,sync status=progress` then enter password of admin_account again)

For me the final command was:

```
sudo dd if=/Users/ben/Downloads/ArkOS.img of=/dev/disk4s1 bs=1m conv=noerror,sync status=progress
```