balena-io/etcher #4375: Echer 1.19.25 "Something went wrong. If it is a compressed image, please check that the archive is not corrupted"

jakobsatoshi, 2024-12-28:
- **Etcher version:** Echer 1.19.25
- **Operating system and architecture:** MacOS Sonoma 14.6.1
- **Image flashed:** tails-amd64-6.10
- **What do you think should have happened:** Nao tenho ideia. Reinstalei o BalenaEtcher, troquei o pen drive, formatei o pendrive no formato ExFat e nada funcionou. 
- **What happened:**Something went wrong. If it is a compressed image, please check that the archive is not corrupted
- **Do you see any meaningful error information in the DevTools?** Starting...
index.js:2 Verifying 1 device, 98% at 19.30 MB/s (total 19.30 MB/s) eta in 1s with 0 failed devices
9index.js:2 Starting...
index.js:2 fail event
index.js:2 Object
index.js:2 Object
2index.js:2 Starting...
index.js:2 Flash results Object
index.js:2 Object
index.js:2 Starting...
index.js:2 Error: The writer process ended unexpectedly
    at i (index.js:353:5238)
    at t.createUserError (index.js:353:6464)
    at m (index.js:140:45502)
    at index.js:140:46208
    at L.<anonymous> (index.js:140:42787)
    at L.emit (node:events:518:28)
    at e.exports.H (index.js:2:2317562)
    at e.exports.emit (node:events:518:28)
    at index.js:2:2293382
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
(anonymous) @ index.js:2
2index.js:2 Starting...
media/icon.png:1 
        
        
       Failed to load resource: net::ERR_FILE_NOT_FOUND
index.js:2 Verifying 1 device, 99% at 19.30 MB/s (total 19.30 MB/s) eta in 0s with 0 failed devices
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 2 events"}
(anonymous) @ index.js:2
index.js:2 CHILD: flash done Object
index.js:2 Flash results Object
2index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 2 events"}
(anonymous) @ index.js:2
index.js:2 Starting...
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
e.warn @ index.js:2
e.handleResponse @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
a @ index.js:355
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 2 events"}
(anonymous) @ index.js:2
e.warn @ index.js:2
e.handleResponse @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
a @ index.js:355
Promise.then (async)
l @ index.js:355
(anonymous) @ index.js:355
h @ index.js:355
e.send @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
(anonymous) @ index.js:355
h @ index.js:355
e.flush @ index.js:2
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
e.warn @ index.js:2
e.handleResponse @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
a @ index.js:355
Promise.then (async)
l @ index.js:355
(anonymous) @ index.js:355
h @ index.js:355
e.send @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
(anonymous) @ index.js:355
h @ index.js:355
e.flush @ index.js:2
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
Promise.then (async)
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 3 events"}
(anonymous) @ index.js:2
e.warn @ index.js:2
e.handleResponse @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
a @ index.js:355
Promise.then (async)
l @ index.js:355
(anonymous) @ index.js:355
h @ index.js:355
e.send @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
(anonymous) @ index.js:355
h @ index.js:355
e.flush @ index.js:2
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
Promise.then (async)
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
Promise.then (async)
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
Promise.then (async)
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
Promise.then (async)
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
2index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count
(anonymous) @ index.js:2
e.error @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
a @ index.js:355
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
e.warn @ index.js:2
e.handleResponse @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
a @ index.js:355
Promise.then (async)
l @ index.js:355
(anonymous) @ index.js:355
h @ index.js:355
e.send @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:2
(anonymous) @ index.js:355
(anonymous) @ index.js:355
(anonymous) @ index.js:355
h @ index.js:355
e.flush @ index.js:2
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
Promise.then (async)
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
Promise.then (async)
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
Promise.then (async)
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
Promise.then (async)
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
Promise.then (async)
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
Promise.then (async)
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
Promise.then (async)
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
Promise.then (async)
(anonymous) @ index.js:2
r @ index.js:2
setTimeout (async)
(anonymous) @ index.js:2
e.schedule @ index.js:2
(anonymous) @ index.js:2
index.js:2

kissu, 2024-12-28:
I made a fork of etcher a few years ago before it got acquired by Balena.
Maybe this version works better than the current one full of premium bloat.
https://github.com/kissu/etcher

jakobsatoshi, 2024-12-29:
Obrigado Kissu, mas eu tentei através do seu link e está dando este aviso: Sun Dec 29 2024 10:08:11 GMT-0300 (-03) Open modal ({"name":"warning","applicationSessionUuid":"099268da-d7b5-4aff-a8b9-c701b47c856e","flashingWorkflowUuid":"c2f3ff0e-183d-4b94-b68a-e244f89ddf86"})
/Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58 {"stack":"Error: EPERM: operation not permitted, open '/Users/felipe/Desktop/tails-amd64-6.10.img'","message":"EPERM: operation not permitted, open '/Users/felipe/Desktop/tails-amd64-6.10.img'","name":"Error","code":"EPERM","syscall":"open","errno":-1}
console.(anonymous function) @ /Applications/balenaEtcher.app/Contents/Resources/app.asar/node_modules/raven-js/dist/raven.js:58

st412m, 2025-02-02:
> I made a fork of etcher a few years ago before it got acquired by Balena. Maybe this version works better than the current one full of premium bloat. https://github.com/kissu/etcher

It`s worked! Thx.