balena-io/etcher #4376: Issue: Error when selecting image file in Etcher v1.19.25

sai8151, 2024-12-28:
### **Issue: Error when selecting image file in Etcher v1.19.25**

#### **Environment Details**  
- **Etcher Version:** v1.19.25 ([[Release](https://github.com/balena-io/etcher/releases/tag/v1.19.25)](https://github.com/balena-io/etcher/releases/tag/v1.19.25))  
- **Operating System:** Zorin OS (Linux)  
  ```
  Linux zorin-HP-Laptop-14-dq2535tu 5.15.0-119-generic #129~20.04.1-Ubuntu SMP Wed Aug 7 13:07:13 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
  ```  
- **Etcher SDK Version:** 9.1.2 ([[Commit Reference](https://github.com/balena-io/etcher/commit/c726b51dca3383c76f4bf824fd5d594ac3069180)](https://github.com/balena-io/etcher/commit/c726b51dca3383c76f4bf824fd5d594ac3069180))  

#### **Steps to Reproduce**  
1. Open Etcher v1.19.25.  
2. Select the ISO file located at `/home/zorin/Downloads/kali-linux-2024.4-live-amd64.iso`.  

#### **Error Message**  
Upon selecting the image file, the following error is displayed:  
```
Something went wrong while opening /home/zorin/Downloads/kali-linux-2024.4-live-amd64.iso

Error: (0, h.requestMetadata) is not a function
```

#### **Expected Behavior**  
The selected ISO file should open without errors.

#### **Actual Behavior**  
The operation fails with the above error.

#### **Additional Information**  
- The issue may be related to the recent patch ([[c726b51](https://github.com/balena-io/etcher/commit/c726b51dca3383c76f4bf824fd5d594ac3069180)](https://github.com/balena-io/etcher/commit/c726b51dca3383c76f4bf824fd5d594ac3069180)), which bumped Etcher SDK to version 9.1.2.  
- The ISO file is valid and has been used with other tools successfully.

#### **Request**  
Could you investigate the issue and provide guidance or a fix? If additional logs or steps are needed, please let me know.

edurenye, 2025-01-29:
I'm still getting the same error when going back to v1.19.21

edurenye, 2025-01-29:
Going back to v1.18.11 fixed it for me

lurch, 2025-01-29:
> `Error: (0, h.requestMetadata) is not a function`

See #4150