balena-io/etcher #4380: Device Not detected to Flash

TCarl12, 2025-01-01:
- **Etcher version:** 1.19.25
- **Operating system and architecture:** Windows
- **Image flashed:** HiveOS installer
- **What do you think should have happened:** Device should have been selectable for flash
- **What happened:**SSD is not detected when trying to Flash. A orange triangle pops up like it is hidden

KERNAMON, 2025-01-13:
Exact same problem for us with Etcher v1.19.25

Operating system and architecture: Windows 10
Image to be flashed: linuxmint-22-cinnamon-64bit.iso

 Phillips USB 32GB flash-drive is not detected when trying to Flash. Not even showing up, despite an orange triangle popping up indicating that main disk "C:" is hidden!?

nonoy79, 2025-03-23:
IM ALSO THE SAME