balena-io/etcher #4386: the etcher just made my pendrive not accessable anymore?

sanwenyu1004, 2025-01-11:
- **Etcher version:** 1.19.25
- **Operating system and architecture:** Windows 11 24H2
- **Image flashed:** lastest ubuntu LTS
- **What do you think should have happened:**  the pendrive should not have been seperated into 2 parts and the rest of the empty drive were deleted and left blank? It should have only been 1 thing like a windows usb booter?
- **What happened:** I was installing and windows started telling me cannot read from usb, you need to format this usb (which cant work because of a writing protection)
- **Do you see any meaningful error information in the DevTools?** index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 3 events"}
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 2 events"}
(anonymous) @ index.js:2
2index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry count
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Warn]: {code: 0, error: "Status 'unknown' provided for 1 events"}
(anonymous) @ index.js:2
index.js:2 Amplitude Logger [Error]: Event rejected due to exceeded retry
something like that

pls help this is my first time installing linux and it has already costed me a new drive

PatoFlamejanteTV, 2025-01-20:
Try seeing the current partitions on **Win + X > Disk Management.**

lurch, 2025-01-21:
See https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos and https://blog.balena.io/did-etcher-break-my-usb-sd-card/

github12101, 2025-02-11:
> pls help this is my first time installing linux and it has already costed me a new drive

Check Windows Journal for read/write errors when attempting to access the stick. If USB stick cannot write, then it is damaged, and you cannot fix it, only throw it away. It's not BalenaEtcher's fault.
You can also try ValiDrive tool (free) from Steve Gibson at grc.com. This tool with verify if stick can read and write properly.

thynus, 2025-02-26:
> See https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos and https://blog.balena.io/did-etcher-break-my-usb-sd-card/

Scroll down to the Windows section and follow the steps.