balena-io/etcher #4396: Something went wrong. If it is a compressed image, please check that the archive is not corrupted"

droslean, 2025-02-06:
version: `1.19.25`

I am trying to flash: https://sourceforge.net/projects/pb-gh-releases/files/S15Pup64_release/ which is an official .iso image.

Error:

[image]

koleto, 2025-02-10:
I get the same message which aways happens when I try to flash large images over 4Gb with partition tables inside. A conventional way with dd work for me in this case or **Pi Imager** as a custom image works too. This is a bug that was introduced some half year ago on etcher for MACOS ARM64.

MichaIng, 2025-02-16:
We faced the same with an MBR partitioned Linux OS image in raw `.img` format, with partition only a little above 1 GiB size. But more strangely, after I tested a few other images, generated with the same build tools, just for different hardware, the originally failing image suddenly flashed without the error.

Is here a way to enable sort of debug mode/logs, to get some more info?

With Rufus, dd etc, it works all well. Also, while the images are xz-compressed, we ran into the same error when uncompressing it first, and selecting the raw `.img` file and when selecting the `.img.xz` archive.

EagerToLearnBrit, 2025-02-17:
All I get is this >>>>>     [image]

pfafman, 2025-02-28:
Having the same issue.  xz compressed image fails to flash but if I uncompress it outside of the app it flashes fine. 
Have seen issue with both 1.19.25 and 2.1.0

The image is of a modified PiOS install.

lurch, 2025-03-03:
@EagerToLearnBrit See #4150

fakhir, 2025-08-20:
I was facing this problem on Linux with Balena Etched v2.1.4 when writing a roughly 15GB .img.xz raw image containing a partition table. I then enabled debugging with "Ctrl+Shift+I" and noticed that it was generating a "Source and destination checksums do not match" when validating the last block on the device. So then I modified my raw image to make it a multiple of 4 MiB (it was previously only a multiple of 512 and not even a multiple of 1024 bytes). After resizing the image, the image flashing was successful.