balena-io/etcher #4399: BalenaEtcher broke my USB

agila5, 2025-02-09:
A few days ago I flashed a Ubuntu image on my new USB stick. Everything went fine. However, now when I check the status of the USB stick it appears to be unusable. 

I followed the tutorial detailed at <https://blog.balena.io/did-etcher-break-my-usb-sd-card/> but with no success. In fact, when I plug in the USB stick and run `list disk` I see the following: 

[image]

and when I select disk 1 and run `clean` I see

[image]

The following is a detail of that disk: 

[image]

Could you please suggest how can I fix my USB? Thanks

github12101, 2025-02-11:
Check Windows Journal for read/write errors when attempting to access the stick. If USB stick cannot write, then it is damaged, and you cannot fix it, only throw it away.
You can also try ValiDrive tool (free) from Steve Gibson at grc.com. This tool with verify if stick can read and write properly.

agila5, 2025-02-16:
ValiDrive didn't even recognise my USB, I guess I just have to throw it away....

closed 2025-02-16 by agila5