balena-io/etcher #4402: Failed to burn SteamOS to MacBook Pro 15.3 system!!!

iMacbooke, 2025-02-15:
Good afternoon! I need to use Apple Mac computer system version 15.3 to burn SteamOS into a USB flash drive. Everything is done according to the steps, but why does the progress bar always stop working after clicking Start Production? (I have tried countless times but failed to burn successfully!!! 0%…..0%!!!)

lurch, 2025-02-16:
IIRC there have been a few other Etcher issues mentioning SteamOS, and I think people said that things work better if you manually unpack the SteamOS image first, before trying to flash it with Etcher?
https://github.com/balena-io/etcher/issues?q=is%3Aissue%20SteamOS

iMacbooke, 2025-02-16:
> IIRC还有其他一些Etcher问题提到SteamOS，我想人们说，如果你先手动解压缩SteamOS图像，然后再尝试用Etcher刷新它，事情会更好？
> https://github.com/balena-io/etcher/issues?q=is%3Aissue%20SteamOS

Thanks a lot, I'll try to fix it myself.

iMacbooke, 2025-02-16:
> > IIRC还有其他一些Etcher问题提到SteamOS，我想人们说，如果你先手动解压缩SteamOS图像，然后再次尝试用Etcher刷新它，事情会更好？
> > https://github.com/balena-io/etcher/issues?q=is%3Aissue%20SteamOS
> 
> 非常感谢，我会试着自己修好它。

It just makes me wonder, because when I ran into this problem a few years ago, I had to burn SteamOS to a USB drive on Windows OS and everything worked fine (I didn't unzip the .bz2, it worked fine on my non-Mac computer). IIRC the case you mentioned might solve my problem.

iMacbooke, 2025-02-16:
> > > IIRC还有其他一些Etcher问题提到SteamOS，我想人们说，如果你先手动解压缩SteamOS图像，然后再次尝试用Etcher刷新它，事情会更好？
> > > https://github.com/balena-io/etcher/issues?q=is%3Aissue%20SteamOS
> > 
> > 
> > 非常感谢，我会试着自己修好它。
> 
> 这让我很好奇，因为几年前我遇到这个问题时，我不得不在Windows操作系统上将SteamOS刻录到USB驱动器上，一切正常（我没有解压缩.bz2，它在我的非Mac计算机上工作正常）。IIRC，你提到的案例可能会解决我的问题。

Hello! I unzipped the bz2 file according to your instructions and it worked. But it seems that the progress bar cannot be completed after Etcher starts to verify (I tried twice but it failed to complete the verification, the first time was 86%, the second time was 64%). Is this correct?

davidt1971, 2025-03-06:
I too can confirm, that the steam deck compressed file MUST be manually extracted now prior to using in etcher on osx. Validation of the written USB went to 100% for me without any issues. Have still to use the flashed image on the steam deck. Thanks.