balena-io/etcher #4408: ******

beatnickgr, 2025-02-21:
I'm trying to install etcher, using the line from their wiki and replacing the ****** with 1.19.25

> sudo apt install ./balena-etcher_1.19.25_amd64.deb
> Reading package lists... Done
> E: Unsupported file ./balena-etcher_1.19.25_amd64.deb given on commandline

The version is 7 characters, if I remove the dots it's 5 characters, so I don't know what format are the 6 asterisks. I've tried i've tried the 6 characters v11925 but it doesn't work either.

Any hint what are the ****** (and maybe update the wiki about it)

I put a message on an existing thread but it doesn't bump the thread, so i make a new thread...

lurch, 2025-02-22:
The number of asterisks isn't significant, it just means "replace the asterisks with whatever version is in the filename you downloaded".
Confusingly, it looks like `apt` gives you that "Unsupported file" error if the filename you give it doesn't actually exist! (which isn't very helpful 🤦 )

So you need to:
1. Download the latest Etcher `.deb` file from https://github.com/balena-io/etcher/releases/
2. `cd` to wherever you downloaded the file to (e.g. `cd ~/Downloads/`)
3. `sudo apt install ./the_filename_you_just_downloaded.deb` should then work

beatnickgr, 2025-02-22:
Thank you lurch, it's working. Would be more intuitive to update the documentation with the 2 dots, and correct number of asterisks.

lurch, 2025-02-22:
The number of asterisks would need to change depending on the Etcher version number, e.g. `balena-etcher_1.19.25_amd64.deb` vs `balena-etcher_2.0.0_amd64.deb`

closed 2025-02-22 by beatnickgr