balena-io/etcher #4412: Corrupts USB installing Linux

SantiagoRj, 2025-02-26:
I have 2 USBs (32GB and 64GB), formatted/fresh, and I wanted to make my USBs bootable to install Linux. I tried with Ubuntu and now with Debian. And with both corrupts my USBs.
I select my .iso, select the USB and "Flash!"/begin.
For Ubuntu I tried 24.04.2 LTS, and for the Debian one I tried with the netinst amd64 one. More precisely yesterday all of this haha
Then give me a error like something went wrong:
[image]

And leave my little USB fucked :(
[image]
Can't formatted, use it, I must reformat it by console.

My PC if helps:
Ryzen 3 3200g
Windows 11 Pro 23H2
RAM 16GB 2400 MT/s
Version app 1.19.25

And I gotta admit, this make me up set as FUCK. Am I doing something wrong or it's a fucking bug/error.

lurch, 2025-02-28:
See https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos and https://blog.balena.io/did-etcher-break-my-usb-sd-card/

SantiagoRj, 2025-07-01:
Hey! I came back haha
I read both sites, I tried with a Fedora .iso, I tried with a brand new USB, I reformat the USB again. And still doesn´t work.
Still happens the same error message, the USB looks "broken"/corrupt (can't access, but appears in the terminal list disk), and absolute NOT boot anything at all when I select it by the BIOS.
So is absolute not an false-positive or it's safe just ignore it, no... doesn´t work either...

I genially have no idea at all. I kinda give up man :(

closed 2025-07-01 by SantiagoRj