balena-io/etcher #4414: Flash problem

njrtnrjh, 2025-03-03:
[image]

Trying to flash Ubuntu 24.04.2 LTS and it never proceeds past this screen. I've tried redownloading the ISO, reinstalling balena etcher, restarting etc and no progress. 
I'm completely new to this, the purpose of installing ubuntu was as a requirement for a JS bootcamp taught in linux so I need a solution in the most basic of terms

njrtnrjh, 2025-03-03:
Not clear from the screenshot but I've selected the ISO under "Flash from file" and the option becomes greyed out and the cursor is locked, it doesn't allow me to select the target usb drive

MarcusPro7, 2025-03-05:
I have the same problem as you, in my case while trying to flash an ISO image of [ClearLinux](https://www.clearlinux.org/downloads.html). 
I’m using Windows 11 24H2 (OS Build 22631.2428) and Balena Etcher 2.1.0.

closed 2025-03-05 by njrtnrjh

MarcusPro7, 2025-03-05:
I just tried using Windows 10 22H2 (OS Build 19045.3803) instead, and it seems to be working.