balena-io/etcher #4422: MacOS: Validation fails if image was compressed

utherp, 2025-03-24:
Version: 2.1.0
Chip: Apple M3 Pro
macOS: 15.3.1

I have tried this with xz, zip, with bmaps and without.  If the image is compressed when I burn it, it will write completely, then immediately fail validation... If I uncompress it first, then it succeeds.

I have confirmed there are several blocks that are different on the image vs the jump drive afterwards.  FYI: it's *always* the same blocks.  I can even rewrite those blocks with dd and they stay correct.

What I suspect is happening is that after releasing the write handle on the block device, the system is rescanning it and doing something to the metadata of file systems, all before you get a chance to start the validation.  

While watching the output of `diskutil activity`, immediately after it finishes the write, I see several events including DiskPeek, DiskAppeared, DiskDescriptionChanged, DiskMountApproval,  and DiskUnmountApproval.  Just before these, I see: "DAIdle (no DADiskRef)".  

This suggests to me that you are closing the handle to /dev/rdiskN, then reopening it for read in order to perform your validations.  This obviously worked well enough before, but it appears something has changed in a recent MacOS release.

..yes, tried several images, several machines and several jump drives.  Though I will confess that all my images contain:

GUID partition scheme with an EFI partition (fat16), a linux ext4 partition and vfat (fat32) filesystem.  It appears to be within the two fat partitions that the validations fail consistently.

d-lyder-okeano, 2025-03-26:
I have encountered the same issue.
With the compressed file, not only the validation fails, but the flashed sd-card is corrupted and not bootable.
If I extract it first, everything is working like before. Tested it on different macs (all on the recent version 15.3.2), different sd-cards and different images.

utherp, 2025-04-03:
The *only* difference between it working and not working is whether the image is compressed.  This has been verified with _many_ jump drives and _many_ images on _many_ macs (M1)