balena-io/etcher #4429: broke usb drive

Cthulhuhpl, 2025-04-04:
I wanted to create a Linux bootable flash drive using balenaEtcher. after installing and validating the files, it asked me to format the flash drive, but I failed and now the flash drive is not available. when I try to open it, it gives an error “Please insert a disk into USB Drive”. when I try to format it, the error “There is no disk in drive”. In disk management, the flash drive is visible as an empty device on which it is impossible to create a new volume

[image]

[image]

[image]

[image]

olton, 2025-04-26:
After creating Ubuntu bootable disk, the disk has become inaccessible.

LewAdvSol-Ian, 2025-04-28:
Join the party, it is still happening after years of issue being reported. I have to re-install my OS now because of this stale project. All usb drive have to manually assigned drive letter, diskpart doesnt work, dism shows all good, yet none of this occurred before attempting to flash an ISO to a thumb drive using this POS. Ok I have vented enough, just angry at the time this "solution" is now wasting of mine.