balena-io/etcher #4431: Cannot open ISO

lordofscripts, 2025-04-05:
I downloaded the Ubuntu Desktop ISO (24.04.2). I am able to mount it from Windows 10.

I downloaded Bakens v2.1 (latest) but no matter which ISO I select from Flash from file, I get this error after a few seconds:

```
Something went wrong while opening 
Filepath here.iso

Error: (0, h.requestMetadata) is not a function.
```

rodoviario, 2025-04-06:
I confirm that it doesn't open isos that my previous version of balena did

I've just downloaded balenaEtcher-2.1.0-x64.AppImane

techieshark, 2025-04-06:
If anyone lands here when using a Mac: this may be due to permissions issues (not having full disk access until user opens system settings and changes that).

see also: https://forums.balena.io/t/error-0-h-requestmetadata-is-not-a-function/371138/10

jeanlucmargot, 2025-04-21:
On MacOS, System Settings -> Privacy & Security -> Full Disk Access -> "+" -> balenaEtcher.app
The balenaEtcher error message is not descriptive and ought to be improved.

lurch, 2025-04-23:
See #4150