balena-io/etcher #4432: balenaetcher keeps installing itself continously after deinstallation...

Pommesmajo74, 2025-04-07:
Hi everybody,

i´m having a  bit of a problem. As said in the title, balena etcher keeps installing itself and i have no idea why. I don´t use it anymore, and i deinstalled it several times. Even deep cleaning and deleting all traces didn´t help. After a few days, it pops up on my screen, properly installed. I have tried to stop it, by generating 0-Bytes files, named balenaetcher.exe, and gave it "read-only" attributes. That seemed to work for a few days longer than before, but today it was there again. I tried Apps like Process Monitor and others, to find out, what process is actually installing it, but i couldn´t find any clear sources. 

So, my question is: Does anybody else have this problem? Is there an known Malware, that installes fake-versions of balena etcher? It looks absolutely trustworthy, but i just don´t want it to take over my computer, nor doing things, that i didn´t allow it to.

Any help is deeply appreciated, thanks so far....

aethernet, 2025-05-09:
This is a very weird problem to have. I don't know of anything in etcher itself that would do that.

Pommesmajo74, 2025-05-09:
It really is a weird Problem. I could not find anything similarly. It has some pretty weird behavior, too. It is very hard to trace, wich process initiated the installation.  Monitoring with 'ProcessMonitor' failed. When I try to open the monitor after it has been installed, it crashes. Even Logs are lacking the information. I'm no professional, but I've done everything I could. I keep posting this just in case someone else gets to this point.