balena-io/etcher #4451: Error: (0 , h.requestMetadata) is not a function in MacOS 13.7.5, not raising "full disk access" permission request

okingfisher, 2025-04-26:
Hi,
This happened in MacOS 13.7.5. Downloaded last Etcher, 2.1.0. When trying to use a Linux Mint image, I got this error:

Something went wrong while opening /Users/xxx/Downloads/linuxmint-22.1-cinnamon-64bit(1).iso.
Error: (0 , h.requestMetadata) is not a function

Checked and found several issues related to this error (none recent).
Since it seemed related to a permission issue, I run with sudo from the terminal, no problems. Once done that, it seems to work when running as a normal user.
I wonder if it happened to more people. Just in case this is a bug or maybe something in my computer prevented it running properly in the first try.

lurch, 2025-04-26:
> I wonder if it happened to more people.

You could say that... 🤣  See #4150

okingfisher, 2025-04-26:
Yup, saw that one, but forgot referencing, sorry. Anyway, older version and other SO. I wonder if this has happened recently and what could be the source for this issue. While it works now... it happened a couple of times more while testing, but not reproducible.

lurch, 2025-04-26:
> I wonder if this has happened recently

Yes, if you scroll to the very bottom of 4150 you can see that it _is_ still affecting people (even in newer versions of Etcher and on different OSes).

lifedaemon-kill, 2025-05-04:
on win 11 if i try to run etcher with adm level see the same

srvprivate, 2026-09-23:
September 23, 2026, MacOS 28.0. Still affecting. That's why I'm here. Tried running a Windows 11 ISO from APR and by default no success. Guys, can we get an update on this? This is a stopping bug. Thanks.

zoltan-ongithub, 2026-09-25:
Still the same issue on Mac.

fre4ki, 2026-09-25:
I'm on MacOS. It is a bug. Version 2.1.6 works..

greybody-g0, 2026-09-26:
Layperson/noob here. Trying to update the software on a digital instrument running on a raspberry pi. [Website](https://docs.critterandguitari.com/Organelle/og_sms2/#71-burning-sd-card-disk-image) says it's done by etching the software onto the microSD using BalenaEtcher. Getting the error listed in topic. Tried permissions tricks, no results.

I'd like to try running an older version of Etcher to see if I can bypass the error using the administrative permission trick that seems to be working for some folks. How do I download an older version of Etcher?

MacOS Tahoe 26.6.2 / BalenaEtcher 2.1.7

Thanks

aerlenbach, 2026-09-27:
Still an issue on 2.1.7 on MacOS.

Kicey, 2026-09-27:
> Still an issue on 2.1.7 on MacOS.

Downgrade to 2.1.6. It works for me. It seems there are some compatible issues in 2.1.7 version.

lurch, 2026-09-27:
> Layperson/noob here. Trying to update the software on a digital instrument running on a raspberry pi. [Website](https://docs.critterandguitari.com/Organelle/og_sms2/#71-burning-sd-card-disk-image) [...]

You might want to try the "Use custom" option of [Raspberry Pi Imager](https://www.raspberrypi.com/software/) 🙂

WillO-K, 2026-09-28:
Can confirm it's an issue for me (2.1.7 on MacOS).

ViniciusMacedoCamara, 2026-09-28:
does not work on both 2.1.7 or 2.1.6 (macos)

greybody-g0, 2026-09-29:
> > Still an issue on 2.1.7 on MacOS.
> 
> Downgrade to 2.1.6. It works for me. It seems there are some compatible issues in 2.1.7 version.

Anyone around who could help me understand how to download a previous version of the software? New to github. Thank you

fsyangg, 2026-09-29:
downgraded to 2.1.5, worked

zbuhler, 2026-09-30:
As a work around I happened to have Rapberry Pi Imager installed which allows you to select a generic ISO and flash and it worked great. (https://www.raspberrypi.com/software/). I love balena etcher though so hope it gets fixed soon.

bschaatsbergen, 2026-10-06:
Also ran into this, using BalenaEtcher 2.16 it worked:

* Go to https://etcher.balena.io/#download-etcher
* Copy the link if you hover over "Download"
* Downgrade to 2.1.6