balena-io/etcher #4452: E: Unsupported file ./balena-etcher_2.1.0_amd64.deb given on commandline

LUVLUVLOVE, 2025-04-27:
someone please help I'm new to this somewhat, and when trying to install on terminal i'm getting this error  E: Unsupported file ./balena-etcher_2.1.0_amd64.deb given on commandline
not sure what to do , i'm use a macbook pro that I turned into a ubuntu OS

KeithIMyers, 2025-04-27:
Are you just trying to run the .deb file? If so, that is incorrect. You
need to install it with something like dpkg (sudo dpkg -i
balena-etcher_2.1.0_amd64.deb)

[image: --]
Keith I Myers
[image: http://]http://KMyers.me
Mobile : (305)-929-3475

lurch, 2025-04-28:
Is your Macbook Pro using Apple Silicon? If so, then it won't run an `amd64` binary.

maxkratz, 2025-05-08:
> Is your Macbook Pro using Apple Silicon? If so, then it won't run an `amd64` binary.

(@LUVLUVLOVE probably uses an Intel-based MacBook because Ubuntu OS does not run well on Apple Silicon chips.)