balena-io/etcher #4453: Outdated Etcher Download Link on Website!

Forandoms, 2025-05-05:
I noticed that your website currently links to an outdated version of Etcher:
https://github.com/balena-io/etcher/releases/download/v1.14.3/balenaEtcher-1.14.3.Setup.exe.

However, when I checked your [GitHub repository](https://github.com/balena-io/etcher/releases), I saw that the latest version is 2.1.0. I tried updating the link manually to:
https://github.com/balena-io/etcher/releases/download/v2.1.0/balenaEtcher-2.1.0.Setup.exe
and it worked correctly.

Just wanted to let you know so you can update the website accordingly.

Thanks for the great work, and have a great day!

Best regards,
Forandoms.

drskullster, 2025-08-25:
Thanks for the report @Forandoms. 

What happens is we have a script on the website that fetches the latest version from Github API before displaying the table with all the download links. If the fetch fails for whatever reason, we fall back on v1.14.3. I will publish a quick fix to fall back on the latest version instead (v2.1.4 as of today) and work on a longer term fix that won't require a manual update of the version number (clearly we've done a bad job keeping on top of that).

drskullster, 2025-08-25:
the fallback version is now set to v2.1.4.