balena-io/etcher #4457: Etcher fails on M4 Mac

spl237, 2025-05-08:
I have a 2024 M4 Mac Mini.

I am running the ARM 64 build of Etcher for MacOS.

With both version 1.19.25 and the latest version 2.1.0, this creates corrupted images when trying to flash a Raspberry Pi OS image to an SD card.

The write appears to work without error, but the verify fails at the end with the message that something went wrong, and the resulting card is filled with garbage.

I have enabled Full Disk Access for Etcher in the Privacy & Security settings - it made no difference.

Am I really the only person who has tried Etcher on an M4 Mac? Because this has been broken for a long time - I first reported it on the forums (with no response) back in November.

joohann, 2025-06-07:
Als with M1 OS 15.5

> I have a 2024 M4 Mac Mini.
>
> I am running the ARM 64 build of Etcher for MacOS.
>
> With both version 1.19.25 and the latest version 2.1.0, this creates corrupted images when trying to flash a Raspberry [...]

macuserguru, 2025-06-08:
Mac - M1 MacMini OS 15.5, balenaEtcher-2.1.2-arm64
Run balenaEtcher from Terminal ... su .... open -a .... (my working user have no admin privileges)
Error Message:
`Something went wrong. If it is a compressed image, please check that the archive is not corrupted.
Error spawning the child process`

Are additional privileges necessary for balenaEtcher?
I couldn't find any requirements in the system settings.

Test this with 
linuxmint-22.1-cinnamon-64bit.iso
and
Manjaro-ARM-mate-generic-23.02.img
too

macuserguru, 2025-06-11:
This works:
Sample Manjaro ARM
`hdiutil convert ...../Downloads/_Linux_BSD/_Manjaro/Manjaro-ARM-xfce-generic-efi-23.02.img -format UDTO -o /....../Downloads/_Linux_BSD/_Manjaro/Manjaro-ARM-xfce-generic-efi-23.02.iso
`
UDTO iso.cdr 
UDRW iso.dmg

Sample Linux Mint
https://nerdblog.steinkopf.net/2020/04/bootfaehigen-usb-stick-aus-einem-iso-file-auf-dem-mac-erstellen/ 

```
diskutil list ..... /dev/disk4
diskutil unmountDisk /dev/disk4

sudo dd if=linuxmint-22.1-cinnamon-64bit.iso of=/dev/disk4 bs=1m

```
PS & OffTopic: 
On MB 13" early 2009 white 5.2, 4GB RAM, 500GB SSD - cant install Linux Mint, but Debian works fine debian-12.11.0-amd64-DVD-1.iso