balena-io/etcher #4458: New Feature - Validate Image

capt-phil, 2025-05-08:
Add a stand alone feature to just validate an Image that is already programed in to a device. I want to be able to verify that a programed SD card has the correct software loaded and that it’s not corrupted. You already do a verification when a new device is programed, just make that verification a stand alone option that can be run at anytime.

lurch, 2025-05-08:
...except that would only work until you boot the SD card for the first time. As soon as you boot the SD card, that modifies the filesystem contents (e.g. resizing partitions, writing logfiles or downloading updates), so any attempt to verify that card would then report it as "corrupted".

Tommekeuh, 2025-05-30:
I have a problem that is kinda similar...I want to flash the jp-orin-nano-sd-card-image.zip file. Balena etcher is flashing the file without a problem but when it starts to validate it gives a problem at the end of the validating process. I have taken a screenshot with some more details. I have formated the micro sd card before I started flashing.

[image]