balena-io/etcher #4459: expressvpn deb error when upgrading balena etcher

arthurlutz, 2025-05-08:
upgrading etcher with [deb-get](https://github.com/wimpysworld/deb-get/) I get : 

``
--2025-05-08 21:50:06--  https://github.com/balena-io/etcher/releases/download/v2.1.2/balena-etcher_2.1.2_amd64.deb
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/45055693/e2be65b8-647b-4987-b749-f77c38aba950?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250508%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250508T195007Z&X-Amz-Expires=300&X-Amz-Signature=518345e0d486741641011bb90120ca2a17bb133d6ac7afdf873d64854e134761&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Dbalena-etcher_2.1.2_amd64.deb&response-content-type=application%2Foctet-stream [following]
--2025-05-08 21:50:07--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/45055693/e2be65b8-647b-4987-b749-f77c38aba950?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20250508%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20250508T195007Z&X-Amz-Expires=300&X-Amz-Signature=518345e0d486741641011bb90120ca2a17bb133d6ac7afdf873d64854e134761&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3Dbalena-etcher_2.1.2_amd64.deb&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 118800616 (113M) [application/octet-stream]
Saving to: ‘/var/cache/deb-get/balena-etcher_2.1.2_amd64.deb’

/var/cache/deb-get/balena-etcher_2.1.2_amd64.de 100%[=====================================================================================================>] 113,30M  79,1MB/s    in 1,4s    

2025-05-08 21:50:09 (79,1 MB/s) - ‘/var/cache/deb-get/balena-etcher_2.1.2_amd64.deb’ saved [118800616/118800616]

  [!] ERROR! Missing required information of website package expressvpn:
                                                                        URL=https://www.expressvpn.works/clients/linux/expressvpn__amd64.deb
                                                                                                                                            VERSION_PUBLISHED=
(Reading database ... 1811569 files and directories currently installed.)                                                                                     (Reading database ... 
Preparing to unpack .../balena-etcher_2.1.2_amd64.deb ...
Unpacking balena-etcher (2.1.2) over (2.1.0) ...
Setting up balena-etcher (2.1.2) ...
chmod: cannot access '/opt/balenaEtcher/chrome-sandbox': No such file or directory
```

What's this expressvpn debian package mentionned in the error ?

lurch, 2025-05-08:
`expressvpn` is listed as one of the packages at https://github.com/wimpysworld/deb-get/blob/main/01-main/README.md so perhaps this is a deb-get problem rather than an Etcher problem?

arthurlutz, 2025-05-09:
you're right, after submitting this issue I found traces of expressvpn in get-deb and filed an issue there, it seems that it's a display problem of things running in parallel https://github.com/wimpysworld/deb-get/issues/1443#issuecomment-2864588338 , sorry for the noise and thanks for taking the time to look into this and respond.

closed 2025-05-09 by arthurlutz