balena-io/etcher #4462: Error opening source

dan74987, 2025-05-11:
All my differents attempts to use balenaEtcher v2.1.2 in Windows 10 to create a Linux Mint boot drivefailed with the message:
Something went wrong while opening
C:\Linux-Mint-Mate\Linuxmint-22.1-cinnamon-64bit.iso.
Error: (0, h.requestMetadata) is not a function

(note the first try was Mint-mate, this note was from my second try with cinnamon)

Thank you for your help

[image]

lurch, 2025-05-11:
Seems to be a duplicate of #4150

jhaston, 2025-05-19:
I experienced the same issue and ran windows compatibility checker on Balena.
Running Windows compatibility trouble-shooter appears to have worked, although I haven't used the newly flashed USB stick yet.
The trouble-shooter automatically selected compatibility mode for Windows 8.

[image]

t3chguru36, 2025-05-24:
**I just created a new issue and this may be my problem too rufus doesnt have issues**